ros-visualization/rviz #1511: [noetic] LaserScan message are cutted off after 1 meter

doisyg, 2020-05-27:
Looks like the following function applies a 1 meter default cuttoff:
https://github.com/ros-visualization/rviz/blob/25f377d40be3987ecff3524f16b5d645bc99bb9a/src/rviz/default_plugin/laser_scan_display.cpp#L84

If I replace it by : 
```
projector_->transformLaserScanToPointCloud(fixed_frame_.toStdString(), *scan, *cloud, *tf, 60.0,
                                               laser_geometry::channel_option::Intensity);
```
the problem disappears (or I guess a 60 meters cuttoff is applied)

doisyg, 2020-05-27:
By default the range_max value of the LaserScan should be used as the cutoff, so maybe a bug here:
https://github.com/ros-perception/laser_geometry/blob/d55d1e297b999b1dd7783665cdc8610b0351a827/include/laser_geometry/laser_geometry.h#L253

rhaschke, 2020-05-27:
Looks like, we missed this when migrating from tf1 to tf2, because the [old tf-based function](https://github.com/ros-perception/laser_geometry/blob/d55d1e297b999b1dd7783665cdc8610b0351a827/include/laser_geometry/laser_geometry.h#L222-L229) isn't available anymore for tf2, but silently shadowed by [another one](https://github.com/ros-perception/laser_geometry/blob/d55d1e297b999b1dd7783665cdc8610b0351a827/include/laser_geometry/laser_geometry.h#L285-L293).
We just need to add the -1.0 default cutoff. Could you file a corresponding PR, please?

doisyg, 2020-05-27:
https://github.com/ros-visualization/rviz/pull/1512

rhaschke, 2020-05-28:
Fixed via #1512

closed 2020-05-28 by rhaschke