ros-visualization/rviz #1513: Noetic: orientation quaternion of /initialpose is all zero

at-wat, 2020-05-29:
Orientation field of `/initialpose` and `/move_base_simple/goal` are all zero for any given poses.
I'll dig into the code probably next week.

For example,
```yaml
$ rostopic echo /initialpose 
header: 
  seq: 0
  stamp: 
    secs: 1590749960
    nsecs: 321122626
  frame_id: "map"
pose: 
  pose: 
    position: 
      x: 0.8154544830322266
      y: 0.20693016052246094
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 0.0
  covariance: [0.25, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.25, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.06853892326654787]
---
$ rostopic echo /move_base_simple/goal 
header: 
  seq: 0
  stamp: 
    secs: 1590749974
    nsecs: 422684186
  frame_id: "map"
pose: 
  position: 
    x: 0.7333250045776367
    y: 0.9347925186157227
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0
---
```

### Your environment
* OS Version: Ubuntu 20.04 (docker)
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    rviz version 1.14.0
    compiled against Qt version 5.12.8
    compiled against OGRE version 1.9.0 (Ghadamon)
    Forcing OpenGl version 0.
    Stereo is NOT SUPPORTED
    OpenGl version: 4.6 (GLSL 4.6).
    ```

rhaschke, 2020-05-29:
This should be fixed via #1510. But, it's not yet released.

closed 2020-05-29 by rhaschke

at-wat, 2020-05-29:
Oh, sorry for duplication.
Happy to hear it has been already fixed.