ros-visualization/rviz #1515: Failed to save rviz config

darrenjkt, 2020-06-10:
Describe your issue here and explain how to reproduce it.

When I change my rviz config, then click file > save config as > h2.rviz

```
    Failed to open /opt/ros/melodic/share/rviz/h2.rviz for writing.
```

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
[ INFO] [1591762597.218020280]: rviz version 1.13.12
[ INFO] [1591762597.218085425]: compiled against Qt version 5.9.5
[ INFO] [1591762597.218093680]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1591762597.221079106]: Forcing OpenGl version 0.
[ INFO] [1591762597.678897836]: Stereo is NOT SUPPORTED
[ INFO] [1591762597.678943111]: OpenGl version: 4.6 (GLSL 4.6).
    ```

rhaschke, 2020-06-10:
Obviously, you don't have permission to write in `/opt/ros/melodic`!

closed 2020-06-10 by rhaschke

wangchen-1994, 2020-11-27:
hi,you can try to change folder to your work space,such as catkin_ws/

AdamArena, 2022-02-15:
Such a monumentally helpful answer @rhaschke. Having encountered the same issue myself and now seeing your descriptive answer I know exactly how to proceed from here! Great!

chanakatb, 2022-07-03:
> hi,you can try to change folder to your work space,such as catkin_ws/

Thanks. It works.