ros-visualization/rviz #1518: 'rviz::VisualizationManager' causes memory leak

turinglife, 2020-06-12:
Describe your issue here and explain how to reproduce it.

"manager_ = new rviz::VisualizationManager( render_panel_ );" will cause memory leak.

reproduce:

1st: run this sample code: https://github.com/ros-visualization/visualization_tutorials/blob/hydro-devel/librviz_tutorial/src/myviz.cpp

2nd: use 'top -p myrviz_pid' to determine whether the 'RES' value has been increasing.

conclusion: ~12 hours, there are around 3MB memory leak. 

### Your environment
* OS Version: e.g. Ubuntu 16.04
* ROS Distro: [Kinetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    Stereo is NOT SUPPORTED
    OpenGl version: 4.6 (GLSL 4.6).
    ```
* If source build, which git commit?

rhaschke, 2020-06-12:
Increasing memory in top isn't a proof for a memory leak. rviz receives (and stores) a lot of messages over time and thus increases its memory usage. Leakages should be detected via [sanitizers](https://github.com/google/sanitizers/wiki/AddressSanitizer) and I removed several real leakages in the past.

closed 2020-06-12 by rhaschke

turinglife, 2020-06-12:
Actually, the memory usage of the host will also have been increasing and I haven't observed any downward trend during the period of around 10 hours.

I agree with your point about top, but it is worth noting whether the continuously increasing trend keeps proceeding in a long-term (~several days or more) experiment.

turinglife, 2020-06-12:
> Increasing memory in top isn't a proof for a memory leak. rviz receives (and stores) a lot of messages over time and thus increases its memory usage. Leakages should be detected via [...]

Actually, the memory usage of the host will also have been increasing and I haven't observed any downward trend during the period of around 10 hours.

I agree with your point about top, but it is worth noting whether the continuously increasing trend keeps proceeding in a long-term (~several days or more) experiment.

rhaschke, 2020-06-12:
Indeed, if memory is continuously increasing over several hours, we should figure out why. However, your issue description doesn't provide any information about the (ROS) context of this increase. In order to fix a potential issue, we need to know where memory is allocated over time (and optionally whether it is released in the end, which doesn't help in your case though).
So, if you can provide more detailed information, feel free to reopen the issue.