ros-visualization/rviz #1522: Segmentation fault when running rviz on wsl2 in Windows10

Erdemhan, 2020-06-28:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 18.04 LTS
* ROS Distro: Melodic

Hi i have a problem very similar to #1438 

I am using wsl2 ubuntu on win10 and VcXsrv as X server i can use rqt_graph etc.

When i run rviz it gives these outputs:

```
[ INFO] [1593348825.014098400]: rviz version 1.13.12
[ INFO] [1593348825.014161600]: compiled against Qt version 5.9.5
[ INFO] [1593348825.014196400]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1593348825.018004200]: Forcing OpenGl version 0.
Segmentation fault
```

and when i try LIBGL_ALWAYS_INDIRECT= as the solution in #1438 

it gives:
```
[ INFO] [1593348979.559661800]: rviz version 1.13.12
[ INFO] [1593348979.559721800]: compiled against Qt version 5.9.5
[ INFO] [1593348979.559765300]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1593348979.564390700]: Forcing OpenGl version 0.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Segmentation fault
```

how can i run rviz in wsl2 ubuntu (18.04 LTS) ?

Thanks in advance.

rhaschke, 2020-06-29:
The errors hint at issues regarding OpenGL usage in wsl2. Can you run any other OpenGL program, e.g. glxgears?

Erdemhan, 2020-06-29:
> 
> 
> The errors hint at issues regarding OpenGL usage in wsl2. Can you run any other OpenGL program, e.g. glxgears?

Without `LIBGL_ALWAYS_INDIRECT=`  :
```
~$ glxgears
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
29315 frames in 6.8 seconds = 4300.342 FPS
17850 frames in 6.0 seconds = 2952.822 FPS
18993 frames in 6.2 seconds = 3070.950 FPS
17980 frames in 6.4 seconds = 2799.104 FPS
```

With  `LIBGL_ALWAYS_INDIRECT=` :
```
~$ glxgears
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
60977 frames in 7.9 seconds = 7676.433 FPS
35410 frames in 13.7 seconds = 2591.643 FPS
37528 frames in 12.5 seconds = 2997.748 FPS
38136 frames in 12.8 seconds = 2971.624 FPS
```

Erdemhan, 2020-06-29:
Solved the problem! with both adding 
`LIBGL_ALWAYS_INDIRECT=` to bashrc and
`-nowgl` to VcXsrv as parameter.

rhaschke, 2020-06-29:
Great.

closed 2020-06-29 by rhaschke

deepseamaster, 2020-09-20:
Thank you

> Solved the problem! with both adding
> `LIBGL_ALWAYS_INDIRECT=` to bashrc and
> `-nowgl` to VcXsrv as parameter.

Thank you so much. I solved the problem.

truthciao, 2020-12-24:
> Solved the problem! with both adding
> `LIBGL_ALWAYS_INDIRECT=` to bashrc and
> `-nowgl` to VcXsrv as parameter.

Exactly same problem solved, thank you.