ros-visualization/rviz #1525: Running with pyside doesn't work

JuliusSustarevas, 2020-07-16:
**Issue:**
I am trying to run rviz via python pyside:
```
import rospy
import sys

setattr(sys, 'SELECT_QT_BINDING', 'pyside')
import rviz

from PySide2 import QtCore, QtGui, QtWidgets
app = QtWidgets.QApplication(sys.argv)
frame=rviz.VisualizationFrame()
frame.setSplashPath("")
frame.initialize()
```

I get:
`>>> import rviz
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rviz/__init__.py", line 17, in <module>
    from . import librviz_shiboken`

Part of this: https://github.com/ros-visualization/visualization_tutorials/issues/51
Part of this: https://github.com/ros-visualization/visualization_tutorials/issues/52

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: [Melodic]
[ INFO] [1594919224.337276138]: rviz version 1.13.13
[ INFO] [1594919224.337304520]: compiled against Qt version 5.9.5
[ INFO] [1594919224.337312176]: compiled against OGRE version 1.9.0 (Ghadamon)

Pyqt4/5 installed
PySide and Pyside2 installed

JuliusSustarevas, 2020-07-16:
Uncommenting this line and building from source seem to have fixed the problem. Still testing. But I suspect the Shiboken lib was just not built  I am  not sure if the reason it was no longer supported is fixed or not. but will try in a bit

# TODO(wjwwood): re-enabled PySide2 support when it is fixed.
add_subdirectory(shiboken)

rhaschke, 2020-07-17:
PySide support was disabled due to https://bugreports.qt.io/browse/PYSIDE-218.
I'm not sure this is fixed. You are welcome to validate the linked test script on a recent version of PySide shipped with Ubuntu 18.04. Note, that PySide is also disabled here (and you should have seen a corresponding cmake info):
https://github.com/ros-visualization/rviz/blob/291df89957e2f643a3fde28e5cb239fe1f6f5b73/src/python_bindings/shiboken/CMakeLists.txt#L14-L23

You might want to try PySide2 instead of PySide1. Not sure this is supported by `python_qt_bindings`.

JuliusSustarevas, 2020-07-17:
Thanks I'll try passing that check too. 

python_qt_binding for melodic talks about pyside2: http://wiki.ros.org/python_qt_binding

I'm not sure how I would make rviz use pyside2 tho, but I guess i need to find where it uses python_qt_binding and somehow setattr to pyside2?

JuliusSustarevas, 2020-07-17:
Mmh. The python_qt_bindings   shiboken_helper  is all setup for Shiboken2 and PySide2. Its is looking for shiboken2 and fails to find it.  Anything that I find online says its suppose to be in /usr/lib/cmake/Shiboke....  But that doesnt exists. 

It uses PYTHON_EXTENSION_SUFFIX: -x86_64-linux-gnu 
to find it. But when I do 
>>locate shiboken2 
that doesnt really come up. Tho to be fair no .cmake files come up anywhere on the system. 

apt show python-pyside2/shiboiken2 
both show installed

I'm pretty clueless here. Trying to learn cmake by doing this :D
Also found this, but probs not related: ttps://bugreports.qt.io/browse/PYSIDE-508

rhaschke, 2020-07-18:
Let's continue the discussion in #1526. Closing here.
By the way, I had issues installing `python-pyside2` in Ubuntu 18.04. There were package requirements, which couldn't be fulfilled.
Hence, I just worked under Ubuntu 20.04.

closed 2020-07-18 by rhaschke

JuliusSustarevas, 2020-07-20:
Awesome, thanks for looking into it. 
Sorry for the dumb question but: 
So I also had pyside2 installation issues with apt-get but pip worked fine and I'm able to use pyside2 under 18.04 to create guis. 
I understand apt-get is a system-wide install from canonical, but if virtual environment is not enabled  an install from pip still makes pyside2 available. But from messing around with it I'm guess pip doesn't install something (I guess cmake files) that is required during rviz build?

rhaschke, 2020-07-20:
The python_qt_binding script searches for [shiboken2](https://github.com/ros-visualization/python_qt_binding/blob/b4d47a73a45a41913770d98f85e9c3a199187bf5/cmake/shiboken_helper.cmake#L18), [PySide2](https://github.com/ros-visualization/python_qt_binding/blob/b4d47a73a45a41913770d98f85e9c3a199187bf5/cmake/shiboken_helper.cmake#L31), and [PythonLibs](https://github.com/ros-visualization/python_qt_binding/blob/b4d47a73a45a41913770d98f85e9c3a199187bf5/cmake/shiboken_helper.cmake#L42).

You should have a look into the cmake log and the file `build/rviz/CMakeCache.txt` to check, whether those were found.
Did you install the python2 or pyhton3 version of PySide? Python2 is required for ROS Melodic.