ros-visualization/rviz #1527: Can't select interactive markers

wolfv, 2020-07-22:
The PR of https://github.com/ros-visualization/rviz/pull/1519 in release 1.13.13 brea makes RViz unable to select interactive marker arrows for me (cf. https://github.com/ros-visualization/interactive_markers/issues/76)

replacing the size==3 case with:

```
      else { // otherwise perform "official" transformation into float-based Ogre::ColourValue and back
        uint32_t pos = (x + y * w) * 3;
        uint8_t* data = (uint8_t*)box.data + pos;
        uint32_t pix_val = 0;
        memcpy((uint8_t*)&pix_val, (uint8_t*)box.data + pos, 3);
        pixels.push_back(colorToHandle(box.format, pix_val));
      }
```

works fine. 

The conversion to float is probably (too) lossy.

rhaschke, 2020-07-22:
Please file a PR if this fixes your issue. Thanks for debugging.

wolfv, 2020-07-22:
I did that: https://github.com/ros-visualization/rviz/pull/1528

Thanks for the awesome RViz work!

rhaschke, 2020-07-22:
Closed via #1528. Sorry, I didn't notice the PR in first place. Thanks for your fix.

closed 2020-07-22 by rhaschke