ros-visualization/rviz #1531: RViz crash when adding Image Display on OSX

Tobias-Fischer, 2020-07-24:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Mac Catalina
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1595566232.356110000]: rviz version 1.14.1
[ INFO] [1595566232.356199000]: compiled against Qt version 5.12.5
[ INFO] [1595566232.356225000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1595566232.366445000]: Forcing OpenGl version 0.
[ INFO] [1595566234.884075000]: Stereo is NOT SUPPORTED
[ INFO] [1595566234.884180000]: OpenGl version: 2.1 (GLSL 1.2).
```

### Stacktrace:
```
Process 6210 stopped
* thread #3, queue = 'OpenGLMT', stop reason = EXC_BAD_ACCESS (code=1, address=0x8)
    frame #0: 0x00007fff4d8de98f libGPUSupportMercury.dylib`gpusLoadCurrentVertexArray + 594
libGPUSupportMercury.dylib`gpusLoadCurrentVertexArray:
->  0x7fff4d8de98f <+594>: movq   0x8(%rbx), %rax
    0x7fff4d8de993 <+598>: movb   (%rax), %al
    0x7fff4d8de995 <+600>: testb  %al, %al
    0x7fff4d8de997 <+602>: je     0x7fff4d8de9bf            ; <+642>
Target 0: (rviz) stopped.
(lldb) bt
* thread #3, queue = 'OpenGLMT', stop reason = EXC_BAD_ACCESS (code=1, address=0x8)
  * frame #0: 0x00007fff4d8de98f libGPUSupportMercury.dylib`gpusLoadCurrentVertexArray + 594
    frame #1: 0x00007fff239f2ca3 AppleIntelBDWGraphicsGLDriver`updateDispatch + 1214
    frame #2: 0x00007fff37749f2b GLEngine`gleDoDrawDispatchCore + 693
    frame #3: 0x00007fff37709a1c GLEngine`glDrawArraysInstanced_STD_Exec + 316
    frame #4: 0x00007fff377098d9 GLEngine`glDrawArrays_UnpackThread + 40
    frame #5: 0x00007fff377459ac GLEngine`gleCmdProcessor + 77
    frame #6: 0x00007fff6446750e libdispatch.dylib`_dispatch_client_callout + 8
    frame #7: 0x00007fff6446cace libdispatch.dylib`_dispatch_lane_serial_drain + 597
    frame #8: 0x00007fff6446d452 libdispatch.dylib`_dispatch_lane_invoke + 363
    frame #9: 0x00007fff64476a9e libdispatch.dylib`_dispatch_workloop_worker_thread + 598
    frame #10: 0x00007fff646c16fc libsystem_pthread.dylib`_pthread_wqthread + 290
    frame #11: 0x00007fff646c0827 libsystem_pthread.dylib`start_wqthread + 15
(lldb) 
```

Hi there,
I built rviz using conda. It starts fine and seems to work fine, however if I try to add an Image it crashes with above error. It also happens with `export LIBGL_ALWAYS_SOFTWARE=1`.

OSX additional output:

```
Process:               rviz [28900]
Path:                  /Users/USER/*/rviz
Identifier:            rviz
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        zsh [28601]
Responsible:           Terminal [28536]
User ID:               501

Date/Time:             2020-07-24 16:45:38.193 +1000
OS Version:            Mac OS X 10.15.3 (19D76)
Report Version:        12
Anonymous UUID:        A85B5C66-06BC-083B-754F-58D9FC7139EE

Sleep/Wake UUID:       955E2BFF-D5F0-4F0C-BF3F-BB30976BB21B

Time Awake Since Boot: 110000 seconds
Time Since Wake:       370 seconds

System Integrity Protection: enabled

[... 594 lines cut ...]
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       377.2M       88 
__OBJC_RO                         32.0M        1 
__OBJC_RW                         1780K        2 
__TEXT                           265.4M      376 
__UNICODE                          564K        1 
mapped file                       57.6M       22 
shared memory                      632K       14 
===========                     =======  ======= 
TOTAL                            949.4M     1265 
TOTAL, minus reserved VM space   948.4M     1265 

Model: MacBookPro12,1, BootROM 189.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2.9 GHz, 16 GB, SMC 2.28f7
Graphics: kHW_IntelIris6100Item, Intel Iris Graphics 6100, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445424533303445422D45474346202020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80CE, 0x4B3445424533303445422D45474346202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.106.3 AirPortDriverBrcmNIC-1440.1)
Bluetooth: Version 7.0.3f5, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

Tobias-Fischer, 2020-07-26:
There is a race condition somewhere. This fixes the problem, but is not a proper fix - it would be great if someone with more rviz / ogre knowledge could give me a hint how a proper fix could look like:

```diff
diff --git a/src/rviz/default_plugin/image_display.cpp b/src/rviz/default_plugin/image_display.cpp
index 3b5d50ce..90252890 100644
--- a/src/rviz/default_plugin/image_display.cpp
+++ b/src/rviz/default_plugin/image_display.cpp
@@ -117,9 +117,9 @@ void ImageDisplay::onInitialize()
   }
 
   render_panel_ = new RenderPanel();
+  render_panel_->getRenderWindow()->addListener(this);
   render_panel_->getRenderWindow()->setAutoUpdated(false);
   render_panel_->getRenderWindow()->setActive(false);
-
   render_panel_->resize(640, 480);
   render_panel_->initialize(img_scene_manager_, context_);
 
@@ -136,12 +136,29 @@ ImageDisplay::~ImageDisplay()
 {
   if (initialized())
   {
+    render_panel_->getRenderWindow()->removeListener(this);
+
     delete render_panel_;
     delete screen_rect_;
     removeAndDestroyChildNode(img_scene_node_->getParentSceneNode(), img_scene_node_);
   }
 }
 
+void ImageDisplay::preRenderTargetUpdate(const Ogre::RenderTargetEvent& /*evt*/)
+{
+  static int count = 0;
+  if(count++>1) {
+    img_scene_node_->setVisible(true);
+  } else{
+    img_scene_node_->setVisible(false);
+  }
+}
+
+void ImageDisplay::postRenderTargetUpdate(const Ogre::RenderTargetEvent& /*evt*/)
+{
+  img_scene_node_->setVisible(false);
+}
+
 void ImageDisplay::onEnable()
 {
   ImageDisplayBase::subscribe();

diff --git a/src/rviz/default_plugin/image_display.h b/src/rviz/default_plugin/image_display.h
index 88492c19..17e4baa8 100644
--- a/src/rviz/default_plugin/image_display.h
+++ b/src/rviz/default_plugin/image_display.h
@@ -59,7 +59,7 @@ namespace rviz
  * \class ImageDisplay
  *
  */
-class ImageDisplay : public ImageDisplayBase
+class ImageDisplay : public ImageDisplayBase, public Ogre::RenderTargetListener
 {
   Q_OBJECT
 public:
@@ -71,6 +71,10 @@ public:
   void update(float wall_dt, float ros_dt) override;
   void reset() override;
 
+  // Overrides from Ogre::RenderTargetListener
+  void preRenderTargetUpdate(const Ogre::RenderTargetEvent& evt) override;
+  void postRenderTargetUpdate(const Ogre::RenderTargetEvent& evt) override;
+
 public Q_SLOTS:
   virtual void updateNormalizeOptions();
 

```

Tobias-Fischer, 2020-07-27:
If I make "count" a (non-static) class attribute, I can add multiple images; otherwise it crashes once adding a second image.

rhaschke, 2020-08-10:
What I am confused about is the fact that the segfault occurs in a thread spawned by OpenGL. Maybe it's related to a broken OpenGL library? Did you try another version of it? @wolfv, what are your experiences?

zzningxp, 2022-06-23:
My issues:

Environment
OS Version: Mac Catalina
ROS Distro: Noetic

Installation:
```
conda create -n ROS python==3.8
conda activate ROS
conda config --add channels conda-forge
conda config --add channels robostack
conda config --set channel_priority strict
conda install ros-noetic-desktop-full
conda install compilers cmake pkg-config make ninja catkin_tools
conda deactivate
conda activate ROS
rosdep init
rosdep update
rviz

```

RViz, Qt, OGRE, OpenGl version as printed by rviz:

```# rviz
[ INFO] [1655954814.977154000]: rviz version 1.14.6
[ INFO] [1655954814.977211000]: compiled against Qt version 5.12.9
[ INFO] [1655954814.977221000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1655954815.017990000]: Forcing OpenGl version 0.
[ INFO] [1655954816.113320000]: Stereo is NOT SUPPORTED
[ INFO] [1655954816.113386000]: OpenGL device: AMD Radeon Pro 560X OpenGL Engine
[ INFO] [1655954816.113403000]: OpenGl version: 2.1 (GLSL 1.2).
[ERROR] [1655954816.339471000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Failed to load library /Users/d/opt/anaconda3/envs/ROS/lib/librviz_default_plugin.dylib. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = dlopen(/Users/d/opt/anaconda3/envs/ROS/lib/librviz_default_plugin.dylib, 9): Library not loaded: @rpath/liburdfdom_model.1.0.dylib
  Referenced from: /Users/d/opt/anaconda3/envs/ROS/lib/liburdf.dylib
  Reason: image not found)
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 1655954817.627902 according to authority unknown_publisher
         at line 278 in /Users/runner/miniforge3/conda-bld/ros-noetic-tf2_1615119151908/work/ros-noetic-tf2/src/work/src/buffer_core.cpp```

Tobias-Fischer, 2022-06-23:
Please open an issue in https://github.com/RoboStack/ros-noetic @zzningxp

rhaschke, 2022-06-23:
Closing here, as this is a RoboStack packaging issue.

closed 2022-06-23 by rhaschke