ros-visualization/rviz #1534: rviz dae file "

JamesNewton, 2020-08-16:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 16.04
* ROS Distro: Kinetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
````
[ INFO] [1597549634.471718284]: rviz version 1.12.17
[ INFO] [1597549634.471928315]: compiled against Qt version 5.5.1
[ INFO] [1597549634.471963392]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1597549634.990012145]: Stereo is NOT SUPPORTED
[ INFO] [1597549634.990159050]: OpenGl version: 3 (GLSL 1.3).
````
### Description

Here is the complete run
````
$ roslaunch urdf_tutorial display.launch model:=/home/jamesn/Documents/Dexter.urdf
... logging to xxxx
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://192.168.0.134:39389/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_gui: True

NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/state_publisher)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[joint_state_publisher-1]: started with pid [2505]
process[robot_state_publisher-2]: started with pid [2506]
process[rviz-3]: started with pid [2507]
[WARN] [1597548587.898550]: The 'use_gui' parameter was specified, which is deprecated.  We'll attempt to find and run the GUI, but if this fails you should install the 'joint_state_publisher_gui' package instead and run that.  This backwards compatibility option will be removed in Noetic.
[ERROR] [1597548589.274097388]: Could not load resource [LowRes_DexterHDIKinematic_Link1_Bottom.dae]: Collada: File came out empty. Something is wrong here.
[ERROR] [1597548589.274555027]: Could not load model 'LowRes_DexterHDIKinematic_Link1_Bottom.dae' for link 'base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource LowRes_DexterHDIKinematic_Link1_Bottom.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1597548589.274796288]: Could not load model 'LowRes_DexterHDIKinematic_Link1_Bottom.dae' for link 'base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource LowRes_DexterHDIKinematic_Link1_Bottom.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
================================================================================REQUIRED process [rviz-3] has died!
process has finished cleanly
log file: /home/jamesn/.ros/log/0aa1d05c-df47-11ea-a068-1c4bd6cf810c/rviz-3*.log
Initiating shutdown!
================================================================================
[rviz-3] killing on exit
[robot_state_publisher-2] killing on exit
[joint_state_publisher-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

````

I've attached the dae file and the urdf file
[LowRes_DexterHDIKinematic_Link1_Bottom.dae.zip](https://github.com/ros-visualization/rviz/files/5079947/LowRes_DexterHDIKinematic_Link1_Bottom.dae.zip)

[Dexter.urdf.zip](https://github.com/ros-visualization/rviz/files/5079946/Dexter.urdf.zip)

rhaschke, 2020-08-16:
I cannot reproduce this issue in the Melodic release of rviz:
[image]
Do you mind updating? Alternatively, use `git bisect` to figure out which commit fixed the issue and I will backport it.

JamesNewton, 2020-08-16:
Please pardon the ROS newbie questions but:
- I don't think I can update past Kinetic on Ubuntu 16.04, but maybe I can update only rviz? Is that going to work?
- bisecting would be actual work, which I have enough of at the moment. I feel bad saying that but... If you tell me it should be ok to update only rviz, I'll update one release at a time and tell you where it starts working. Hopefully that will narrow it down.

rhaschke, 2020-08-17:
Yes, I'm talking about source builds. Just clone rviz into your private ROS overlay workspace and build. All packages in Kinetic and later build with catkin. Building the latest meloic-devel branch of rviz should work on Kinetic as well.

JamesNewton, 2020-08-17:
Followed the workspace overlaying tutorial. 

Cloned rviz at current default (which is melodic-devel right? yeah, `git status` says `On branch melodic-devel`) and catkin-make says a lot, but ends up with this:
````
[ 16%] Building CXX object rviz/src/rviz/CMakeFiles/rviz.dir/frame_manager.cpp.o
[ 16%] Built target rviz_default_plugin_automoc
In file included from /home/jamesn/overlay_ws/src/rviz/src/rviz/frame_manager.cpp:30:0:
/home/jamesn/overlay_ws/src/rviz/src/rviz/frame_manager.h: In member function ‘const std::shared_ptr<tf2_ros::Buffer> rviz::FrameManager::getTF2BufferPtr()’:
/home/jamesn/overlay_ws/src/rviz/src/rviz/frame_manager.h:259:17: error: ‘class tf::TransformListener’ has no member named ‘getTF2BufferPtr’
     return tf_->getTF2BufferPtr();
                 ^
rviz/src/rviz/CMakeFiles/rviz.dir/build.make:302: recipe for target 'rviz/src/rviz/CMakeFiles/rviz.dir/frame_manager.cpp.o' failed
make[2]: *** [rviz/src/rviz/CMakeFiles/rviz.dir/frame_manager.cpp.o] Error 1
CMakeFiles/Makefile2:3318: recipe for target 'rviz/src/rviz/CMakeFiles/rviz.dir/all' failed
make[1]: *** [rviz/src/rviz/CMakeFiles/rviz.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
````

JamesNewton, 2020-08-17:
Did some binary search on what tag versions will build under Kinetic on Ubuntu 16.04: Was NOT able to build 1.13.4. Was able to build 1.12.17, but it still reports the exact same problem.

rhaschke, 2020-08-18:
Sorry, I forgot about the tf2 migration issue. You also need to clone a recent tf2 version into your catkin workspace and build it together with rviz:
`git clone --branch=melodic-devel https://github.com/ros/geometry2`

JamesNewton, 2020-08-18:
````
CMake Error at geometry2/tf2_bullet/CMakeLists.txt:11 (find_package):
  By not providing "Findbullet.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bullet", but
  CMake did not find one.
```
As per:
https://www.programmersought.com/article/7841263710/
`sudo apt-get install libbullet-dev`
Seems better..

rhaschke, 2020-08-19:
Hm. This is strange. Looks like the headers are found in /opt/ros first... Need to investigate myself...
Note, you can easily install all upstream dependencies for your `src` folder via:
`rosdep install --from-paths src --ignore-src --rosdistro $ROS_DISTRO`

rhaschke, 2020-08-19:
Sorry, my fault: I was referring you to the wrong TF version. You need to clone TF1 into your catkin workspace:
`git clone --branch=melodic-devel https://github.com/ros/geometry`

JamesNewton, 2020-08-19:
Removed geometry2 and cloned geometry as indicated. 

````
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 6 packages in topological order:
-- ~~  - geometry (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'geometry' contains non standard content. Use the content of the following file instead: /home/jamesn/overlay_ws/build/catkin_generated/metapackages/geometry/CMakeLists.txt
-- ~~  - eigen_conversions
-- ~~  - kdl_conversions
-- ~~  - tf
-- ~~  - tf_conversions
-- ~~  - rviz
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)
````
I'm not sure how to "Use the content of the following file instead: /home/jamesn/overlay_ws/build/catkin_generated/metapackages/geometry/CMakeLists.txt"

Trying "catkin_make_isolated"

rhaschke, 2020-08-19:
Does catkin provide details on which package is non-catkin? I'm usually building using [catkin tools](https://catkin-tools.readthedocs.io/en/latest/).

JamesNewton, 2020-08-19:
Yeah, it says in the output I posted above: "The CMakeLists.txt of the metapackage 'geometry' contains non standard content." Anyway, individual build might have fixed that, but I'm not sure what's going on with the version. See above.

rhaschke, 2020-08-19:
What's the content of your `ROS_PACKAGE_PATH`? Is your local rviz devel folder listed before /opt/ros?
If not, rebuild from scratch. Something went wrong..

JamesNewton, 2020-08-20:
Ah... I should have `source ~/overlay_ws/devel_isolated/setup.bash`. And then I got confused for a while because 1.13.13 is the melodic top, and the 1.14 must be neodic. 

Anyway, it built and ran! Woo hooo!!!!

And I have /exactly/ the same error with 1.13.13:

````
 roslaunch urdf_tutorial display.launch model:=/home/jamesn/Documents/Dexter.urdf
... logging to /home/jamesn/.ros/log/f8d14f3c-e29e-11ea-a138-1c4bd6cf810c/roslaunch-jamesn-G73Sw2-11307.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://jamesn-G73Sw2:42801/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_gui: True

NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [11323]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f8d14f3c-e29e-11ea-a138-1c4bd6cf810c
process[rosout-1]: started with pid [11336]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [11343]
process[robot_state_publisher-3]: started with pid [11354]
process[rviz-4]: started with pid [11356]
[WARN] [1597898301.933789]: The 'use_gui' parameter was specified, which is deprecated.  We'll attempt to find and run the GUI, but if this fails you should install the 'joint_state_publisher_gui' package instead and run that.  This backwards compatibility option will be removed in Noetic.
[ERROR] [1597898303.941354921]: Could not load resource [LowRes_DexterHDIKinematic_Link1_Bottom.dae]: Collada: File came out empty. Something is wrong here.
[ERROR] [1597898303.958299646]: Could not load model 'LowRes_DexterHDIKinematic_Link1_Bottom.dae' for link 'base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource LowRes_DexterHDIKinematic_Link1_Bottom.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1597898303.958730460]: Could not load model 'LowRes_DexterHDIKinematic_Link1_Bottom.dae' for link 'base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource LowRes_DexterHDIKinematic_Link1_Bottom.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
````

rhaschke, 2020-08-20:
The error message `Collada: File came out empty. Something is wrong here.` is generated by assimp, which is used to parse those files. Looks like the issue is in `libassimp`. Try to update this one. Closing here as this seems to be an upstream issue.

closed 2020-08-20 by rhaschke

JamesNewton, 2020-08-20:
So apparently this has been a known bug in assimp since 2017... _and is still open_? How are you getting that file to work on your computer if it's still a known issue? @rhaschke may I ask what version of the OS you are running, since the issue seems less related to ROS version and more related to OS / assimp.
https://github.com/assimp/assimp/issues/1278

JamesNewton, 2020-08-20:
The entire issue here has been that the CAD files were not being found.

The urdf file was in the ~/Documents folder, with the CAD files, but despite the documentation indicating that the mesh file "must be a local file." this does not (apparently) mean local to the urdf file. As soon as I put it in a package path, the file was loaded and displayed. 

The lack of a package: prefix is clear in the urdf file I attached to the original report.

The error messages do NOT indicate that the file is not found. They say "File came out empty." or "Cannot locate resource"

I would recommend better error messages.

rhaschke, 2020-08-21:
Again, that's an issue of upstream `assimp`. In my newer version (4.1 from stock Ubuntu 18.04), I get the following error:
```
[ERROR] ros.rviz: Could not load resource [LowRes_DexterHDIKinematic_Link1_Bottom.dae]: Unable to open file "LowRes_DexterHDIKinematic_Link1_Bottom.dae".
```