ros-visualization/rviz #1542: rviz crash nav_msgs Path

iljastas, 2020-09-10:
Hi guys,

I'm always get a crash with a nav_mgs::Path:
```

Thread 1 "rviz" received signal SIGFPE, Arithmetic exception.
0x00007ffff7abd79c in rviz::BillboardLine::setupChains() ()
   from /opt/ros/melodic/lib/librviz.so
(gdb) bt
#0  0x00007ffff7abd79c in rviz::BillboardLine::setupChains() ()
    at /opt/ros/melodic/lib/librviz.so
#1  0x00007fffa7c1b175 in rviz::PathDisplay::processMessage(boost::shared_ptr<nav_msgs::Path_<std::allocator<void> > const> const&) ()
    at /opt/ros/melodic/lib/librviz_default_plugin.so
#2  0x00007fffa7c1e4e3 in rviz::MessageFilterDisplay<nav_msgs::Path_<std::allocator<void> > >::incomingMessage(boost::shared_ptr<nav_msgs::Path_<std::allocator<void> > const> const&) () at /opt/ros/melodic/lib/librviz_default_plugin.so
#3  0x00007fffa7c1f6ae in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<nav_msgs::Path_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<nav_msgs::Path_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<nav_msgs::Path_<std::allocator<void> > const>) ()
    at /opt/ros/melodic/lib/librviz_default_plugin.so
#4  0x00007fffa7c24af0 in message_filters::CallbackHelper1T<boost::shared_ptr<nav_msgs::Path_<std::allocator<void> > const> const&, nav_msgs::Path_<std::allocator<void> > >::call(ros::MessageEvent<nav_msgs::Path_<std::allocator<void> > const> const&, bool) () at /opt/ros/melodic/lib/librviz_default_plugin.so
#5  0x00007fffa7c23bcf in message_filters::Signal1<nav_msgs::Path_<std::allocator<void> > >::call(ros::MessageEvent<nav_msgs::Path_<std::allocator<void> > const> const&) () at /opt/ros/melodic/lib/librviz_default_plugin.so
#6  0x00007fffa7c28f51 in tf2_ros::MessageFilter<nav_msgs::Path_<std::allocator<void> > >::CBQueueCallback::call() ()
---Type <return> to continue, or q <return> to quit---
    at /opt/ros/melodic/lib/librviz_default_plugin.so
#7  0x00007ffff41a98a9 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () at /opt/ros/melodic/lib/libroscpp.so
#8  0x00007ffff41ab61b in ros::CallbackQueue::callAvailable(ros::WallDuration) () at /opt/ros/melodic/lib/libroscpp.so
#9  0x00007ffff41eb47e in ros::spinOnce() ()
    at /opt/ros/melodic/lib/libroscpp.so
#10 0x00007ffff7b48e88 in rviz::VisualizationManager::onUpdate() ()
    at /opt/ros/melodic/lib/librviz.so
#11 0x00007ffff6cb7555 in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff6cc3fc7 in QTimer::timeout(QTimer::QPrivateSignal) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff6cc4328 in QTimer::timerEvent(QTimerEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff6cb807b in QObject::event(QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff729d83c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff72a5104 in QApplication::notify(QObject*, QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff6c888d8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
---Type <return> to continue, or q <return> to quit---
#18 0x00007ffff6ce0d2e in QTimerInfoList::activateTimers() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff6ce14f1 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff0025417 in g_main_context_dispatch ()
    at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff0025650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff00256dc in g_main_context_iteration ()
    at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff6ce188f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff6c8690a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007ffff6c8f9b4 in QCoreApplication::exec() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x0000555555554dcf in main ()

```

Does anybody have an advice?

rhaschke, 2020-09-10:
Probably your messages are ill-formed, such that a division by zero (or the like) is happening?

rhaschke, 2020-10-19:
Were you able to fix your issue?

iljastas, 2020-10-19:
Indeed, the message was ill-formed. Thanks for your comment. There were a few nan's. I initialized an array with np.empty...

closed 2020-10-19 by rhaschke