ros-visualization/rviz #1543: *** buffer overflow detected *** RVIZ terminated

cblesing, 2020-09-10:
Hi all, since a few days i get an strange error when using rviz. If i playback a bagfile and want so visualize the laserscans rviz crashes after some time (2-3sec) with the error message down below. Funny thing is that I can display maps taken by slam. As soon as I want to display tf or scans rviz dies. Can somebody tell me if this is a problem related to rviz? 

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

```
 rosrun rviz rviz -l
[ INFO] [1599730078.339555047]: rviz version 1.14.1
[ INFO] [1599730078.339599805]: compiled against Qt version 5.12.8
[ INFO] [1599730078.339612873]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1599730078.346330151]: Forcing OpenGl version 0.
[ INFO] [1599730078.450978874]: Creating resource group General
[ INFO] [1599730078.451022320]: Creating resource group Internal
[ INFO] [1599730078.451036718]: Creating resource group Autodetect
[ INFO] [1599730078.451082368]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[ INFO] [1599730078.451213048]: Registering ResourceManager for type Material
[ INFO] [1599730078.451233262]: Registering ResourceManager for type Mesh
[ INFO] [1599730078.451247259]: Registering ResourceManager for type Skeleton
[ INFO] [1599730078.451293843]: MovableObjectFactory for type 'ParticleSystem' registered.
[ INFO] [1599730078.451357704]: ArchiveFactory for archive type FileSystem registered.
[ INFO] [1599730078.451404472]: ArchiveFactory for archive type Zip registered.
[ INFO] [1599730078.451416786]: ArchiveFactory for archive type EmbeddedZip registered.
[ INFO] [1599730078.451428770]: DDS codec registering
[ INFO] [1599730078.451447783]: FreeImage version: 3.18.0
[ INFO] [1599730078.451460431]: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
[ INFO] [1599730078.451559335]: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
[ INFO] [1599730078.451574145]: PVRTC codec registering
[ INFO] [1599730078.451587327]: ETC codec registering
[ INFO] [1599730078.451599855]: Registering ResourceManager for type HighLevelGpuProgram
[ INFO] [1599730078.451616230]: Registering ResourceManager for type Compositor
[ INFO] [1599730078.451749300]: MovableObjectFactory for type 'Entity' registered.
[ INFO] [1599730078.451766714]: MovableObjectFactory for type 'Light' registered.
[ INFO] [1599730078.451778983]: MovableObjectFactory for type 'BillboardSet' registered.
[ INFO] [1599730078.451793525]: MovableObjectFactory for type 'ManualObject' registered.
[ INFO] [1599730078.451810942]: MovableObjectFactory for type 'BillboardChain' registered.
[ INFO] [1599730078.451826418]: MovableObjectFactory for type 'RibbonTrail' registered.
[ INFO] [1599730078.451870944]: *-*-* OGRE Initialising
[ INFO] [1599730078.451900439]: *-*-* Version 1.9.0 (Ghadamon)
[ INFO] [1599730078.451946937]: OverlayElementFactory for type Panel registered.
[ INFO] [1599730078.451961972]: OverlayElementFactory for type BorderPanel registered.
[ INFO] [1599730078.451976363]: OverlayElementFactory for type TextArea registered.
[ INFO] [1599730078.451992692]: Registering ResourceManager for type Font
[ INFO] [1599730078.452013299]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL
[ INFO] [1599730078.453194094]: Installing plugin: GL RenderSystem
[ INFO] [1599730078.453217833]: OpenGL Rendering Subsystem created.
[ INFO] [1599730078.514729389]: Plugin successfully installed
[ INFO] [1599730078.514783241]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_OctreeSceneManager
[ INFO] [1599730078.515450419]: Installing plugin: Octree Scene Manager
[ INFO] [1599730078.515473302]: Plugin successfully installed
[ INFO] [1599730078.515488561]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_ParticleFX
[ INFO] [1599730078.516263340]: Installing plugin: ParticleFX
[ INFO] [1599730078.516287312]: Particle Emitter Type 'Point' registered
[ INFO] [1599730078.516302172]: Particle Emitter Type 'Box' registered
[ INFO] [1599730078.516316290]: Particle Emitter Type 'Ellipsoid' registered
[ INFO] [1599730078.516329747]: Particle Emitter Type 'Cylinder' registered
[ INFO] [1599730078.516344625]: Particle Emitter Type 'Ring' registered
[ INFO] [1599730078.516359177]: Particle Emitter Type 'HollowEllipsoid' registered
[ INFO] [1599730078.516371956]: Particle Affector Type 'LinearForce' registered
[ INFO] [1599730078.516384941]: Particle Affector Type 'ColourFader' registered
[ INFO] [1599730078.516397796]: Particle Affector Type 'ColourFader2' registered
[ INFO] [1599730078.516411342]: Particle Affector Type 'ColourImage' registered
[ INFO] [1599730078.516429199]: Particle Affector Type 'ColourInterpolator' registered
[ INFO] [1599730078.516444325]: Particle Affector Type 'Scaler' registered
[ INFO] [1599730078.516457930]: Particle Affector Type 'Rotator' registered
[ INFO] [1599730078.516472786]: Particle Affector Type 'DirectionRandomiser' registered
[ INFO] [1599730078.516486831]: Particle Affector Type 'DeflectorPlane' registered
[ INFO] [1599730078.516500936]: Plugin successfully installed
[ INFO] [1599730078.516525703]: CPU Identifier & Features
[... 904 lines cut ...]
0:38(2): error: value of type vec4 cannot be assigned to variable of type error

[ INFO] [1599730109.938607135]: GLSL compile log: rviz/glsl150/billboard.geom0:25(3): error: redeclaration of gl_PerVertex must be a subset of the built-in members of gl_PerVertex
0:27(1): error: identifier `gl_TexCoord' uses reserved `gl_' prefix
0:38(2): error: `gl_FrontColor' undeclared
0:38(2): error: value of type vec4 cannot be assigned to variable of type error

[ INFO] [1599730109.970327501]: GLSL compile log: rviz/glsl150/billboard.geom0:25(3): error: redeclaration of gl_PerVertex must be a subset of the built-in members of gl_PerVertex
0:27(1): error: identifier `gl_TexCoord' uses reserved `gl_' prefix
0:38(2): error: `gl_FrontColor' undeclared
0:38(2): error: value of type vec4 cannot be assigned to variable of type error

[ INFO] [1599730110.001001573]: GLSL compile log: rviz/glsl150/billboard.geom0:25(3): error: redeclaration of gl_PerVertex must be a subset of the built-in members of gl_PerVertex
0:27(1): error: identifier `gl_TexCoord' uses reserved `gl_' prefix
0:38(2): error: `gl_FrontColor' undeclared
0:38(2): error: value of type vec4 cannot be assigned to variable of type error

[ INFO] [1599730110.036573089]: GLSL compile log: rviz/glsl150/billboard.geom0:25(3): error: redeclaration of gl_PerVertex must be a subset of the built-in members of gl_PerVertex
0:27(1): error: identifier `gl_TexCoord' uses reserved `gl_' prefix
0:38(2): error: `gl_FrontColor' undeclared
0:38(2): error: value of type vec4 cannot be assigned to variable of type error

*** buffer overflow detected ***: terminated
Aborted (core dumped)
```

cblesing, 2020-09-10:
i fixed it by myself. The buffer overflow was caused by an update from the geometry2 package. 

See: https://github.com/ros/geometry2/issues/480

We can close this issue here

closed 2020-09-10 by rhaschke

nilsbore, 2020-09-16:
@cblesing Could you detail how you solved it? Is it sufficient to recompile a recent geometry2 or does rviz need to be recompiled with the new version? Thanks.