ros-visualization/rviz #1544: LaserScan (possibly point clouds in general) visuals are broken

v4hn, 2020-09-10:
Disclaimers:
- I did not track this down but just want to raise the issue for now
- It is apparently not relevant on at least Ubuntu 18.04 and ROS melodic yet.
- It does not seem to be related to any change in RViz but to some changes in system libraries
- it seems to be reproducible on very current Linux distributions (verified on Debian sid and Lunar-Linux)
- I get the same results with mesa-lib 20.0.4 and 21.1.2, so if it should be related to changes in mesa, it was introduced before then. Ubuntu focal also builds on 20.0.4, so in that case the problem might be reproducible there (and thus on noetic).

When I visualize a LaserScan on these systems and set the style to `Points` of any size, the output looks like this:

[image]

If I switch to **any** other style and no matter what size I specify in the `Size` field, the output looks like this instead:

[image]

Both are obviously broken.

jspricke, 2020-09-10:
Both work fine on Ubuntu 20/Noetic for me

rhaschke, 2020-09-10:
I think this is a duplicate of https://github.com/ros-visualization/rviz/issues/1508. Unfortunately, I don't have a solution for this OpenGL-related issue. It was introduced by the upgrade to Mesa OpenGL 20.x. A workaround mentioned in #1508 is to enforce a lower OpenGL version via `rviz --opengl 210`. Nvidia's OpenGL implementation is not affected.

v4hn, 2020-09-10:
Definitely a duplicate. Closing, sorry for the noise. Hopefully someone with the time or experience can fix this soon.

closed 2020-09-10 by v4hn