ros-visualization/rviz #1545: Displays panel content scrolls horizontally, hides unfold arrows

felixvd, 2020-09-23:
The content in the Displays panel can move horizontally, so that the unfold arrow on the left is hidden, like in these screenshots:  
(In the bottom one, I scrolled back to the left using the arrow keys)

[image]

[image]

In [this MoveIt PR](https://github.com/ros-planning/moveit/pull/2126), a simple fix for a similar problem was proposed, which simply removes the AutoScroll property from the QtCreator .ui file. I would have submitted a PR, but I didn't find the corresponding location in the Rviz repository on first look.

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic

rhaschke, 2020-09-23:
Applying the "simple fix" from MoveIt in rviz's [`PropertyTreeWidget`](https://github.com/ros-visualization/rviz/blob/2b31dc4be241e3ebd3a92c0f94d56fdb71391d51/src/rviz/properties/property_tree_widget.cpp#L61-L79) solves the scrolling issue. However, without scrolling editor widgets, e.g. the button of a combo box, will be hidden behind the scrollbar. Looks like this is a bug in Qt's TreeView, not correctly adapting the visible area when a scrollbar is displayed.
I can imagine two fixes:
- Automatically scroll back as soon as the active editor is closed
- Find a workaround to fix the Qt bug

The latter approach obviously will require more time. First of all, we should try to confirm that it is a Qt bug and report it upstream.

closed 2020-10-18 by rhaschke

felixvd, 2020-10-19:
🎉

commit a3f824fa (2020-10-18): SplitterHandle: Consider scrollbar width

If there is a scrollbar visible, the visible width of the viewport is reduced,
which should be considered when computing the width of the second column.
Without this, the viewport will scroll horizontally to bring the 2nd column into view,
thus hiding the unfold arrows in the first column. Fixes #1545.