ros-visualization/rviz #1547: Hang when loading .rviz with moveit plugin when moveit is not running

rcywongaa, 2020-10-12:
When loading a `.rviz` file that contains the moveit plugin (`moveit_rviz_plugin/MotionPlanning`) but with moveit NOT running, rviz hangs indefinitely until moveit is started.

While hanging, rviz cannot be stopped by SIGINT, only responding to SIGTERM

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

```
[ INFO] [1602480869.072261788]: rviz version 1.13.13
[ INFO] [1602480869.072306395]: compiled against Qt version 5.9.5
[ INFO] [1602480869.072316567]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1602480869.077002382]: Forcing OpenGl version 0.
[DEBUG] [1602480869.077506834]: Creating ClassLoader, base = rviz::Panel, address = 0x5606d985d690
[DEBUG] [1602480869.092832797]: Entering determineAvailableClasses()...
[DEBUG] [1602480869.092860737]: Processing xml file /opt/ros/melodic/share/rviz_visual_tools/plugin_description.xml...
[DEBUG] [1602480869.092966841]: XML file specifies lookup name (i.e. magic name) = rviz_visual_tools/RvizVisualToolsGui.
[DEBUG] [1602480869.092982668]: XML file specifies lookup name (i.e. magic name) = rviz_visual_tools/KeyTool.
[DEBUG] [1602480869.092990586]: Processing xml file /opt/ros/melodic/share/rviz_plugin_tutorials/plugin_description.xml...
[DEBUG] [1602480869.093047065]: XML file specifies lookup name (i.e. magic name) = rviz_plugin_tutorials/Teleop.
[DEBUG] [1602480869.093074495]: XML file specifies lookup name (i.e. magic name) = rviz_plugin_tutorials/Imu.
[DEBUG] [1602480869.093081063]: XML file specifies lookup name (i.e. magic name) = rviz_plugin_tutorials/PlantFlag.
[DEBUG] [1602480869.093087835]: Processing xml file /opt/ros/melodic/share/moveit_ros_visualization/robot_state_rviz_plugin_description.xml...
[DEBUG] [1602480869.093170437]: XML file specifies lookup name (i.e. magic name) = moveit_rviz_plugin/RobotState.
[DEBUG] [1602480869.093179824]: Processing xml file /opt/ros/melodic/share/moveit_ros_visualization/planning_scene_rviz_plugin_description.xml...
[DEBUG] [1602480869.093223099]: XML file specifies lookup name (i.e. magic name) = moveit_rviz_plugin/PlanningScene.
[DEBUG] [1602480869.093259308]: Processing xml file /opt/ros/melodic/share/moveit_ros_visualization/motion_planning_rviz_plugin_description.xml...
[DEBUG] [1602480869.093307200]: XML file specifies lookup name (i.e. magic name) = moveit_rviz_plugin/MotionPlanning.
[DEBUG] [1602480869.093317222]: Processing xml file /opt/ros/melodic/share/moveit_ros_visualization/trajectory_rviz_plugin_description.xml...
[DEBUG] [1602480869.093396370]: XML file specifies lookup name (i.e. magic name) = moveit_rviz_plugin/Trajectory.
[DEBUG] [1602480869.093434026]: Processing xml file /opt/ros/melodic/share/rviz/plugin_description.xml...
[DEBUG] [1602480869.093665958]: XML file specifies lookup name (i.e. magic name) = rviz/Axes.
[DEBUG] [1602480869.093676191]: XML file specifies lookup name (i.e. magic name) = rviz/DepthCloud.
[DEBUG] [1602480869.093684371]: XML file specifies lookup name (i.e. magic name) = rviz/Effort.
[DEBUG] [1602480869.093693900]: XML file specifies lookup name (i.e. magic name) = rviz/FluidPressure.
[DEBUG] [1602480869.093703526]: XML file specifies lookup name (i.e. magic name) = rviz/Grid.
[DEBUG] [1602480869.093735407]: XML file specifies lookup name (i.e. magic name) = rviz/GridCells.
[DEBUG] [1602480869.093760763]: XML file specifies lookup name (i.e. magic name) = rviz/Illuminance.
[DEBUG] [1602480869.093766719]: XML file specifies lookup name (i.e. magic name) = rviz/Image.
[DEBUG] [1602480869.093792908]: XML file specifies lookup name (i.e. magic name) = rviz/InteractiveMarkers.
[DEBUG] [1602480869.093802097]: XML file specifies lookup name (i.e. magic name) = rviz/LaserScan.
[DEBUG] [1602480869.093809475]: XML file specifies lookup name (i.e. magic name) = rviz/Map.
[DEBUG] [1602480869.093815842]: XML file specifies lookup name (i.e. magic name) = rviz/Marker.
[DEBUG] [1602480869.093823564]: XML file specifies lookup name (i.e. magic name) = rviz/MarkerArray.
[DEBUG] [1602480869.093829701]: XML file specifies lookup name (i.e. magic name) = rviz/Odometry.
[DEBUG] [1602480869.093835887]: XML file specifies lookup name (i.e. magic name) = rviz/Path.
[DEBUG] [1602480869.093843159]: XML file specifies lookup name (i.e. magic name) = rviz/PointStamped.
[DEBUG] [1602480869.093850081]: XML file specifies lookup name (i.e. magic name) = rviz/PointCloud.
[DEBUG] [1602480869.093858624]: XML file specifies lookup name (i.e. magic name) = rviz/PointCloud2.
[DEBUG] [1602480869.093868301]: XML file specifies lookup name (i.e. magic name) = rviz/Polygon.
[DEBUG] [1602480869.093875225]: XML file specifies lookup name (i.e. magic name) = rviz/Pose.
[DEBUG] [1602480869.093884032]: XML file specifies lookup name (i.e. magic name) = rviz/PoseArray.
[DEBUG] [1602480869.093892508]: XML file specifies lookup name (i.e. magic name) = rviz/PoseWithCovariance.
[DEBUG] [1602480869.093902081]: XML file specifies lookup name (i.e. magic name) = rviz/Range.
[DEBUG] [1602480869.093910631]: XML file specifies lookup name (i.e. magic name) = rviz/RelativeHumidity.
[DEBUG] [1602480869.093919499]: XML file specifies lookup name (i.e. magic name) = rviz/RobotModel.
[DEBUG] [1602480869.093929020]: XML file specifies lookup name (i.e. magic name) = rviz/Camera.
[DEBUG] [1602480869.093936796]: XML file specifies lookup name (i.e. magic name) = rviz/XYZ.
[DEBUG] [1602480869.093944015]: XML file specifies lookup name (i.e. magic name) = rviz/Intensity.
[DEBUG] [1602480869.093953110]: XML file specifies lookup name (i.e. magic name) = rviz/RGB8.
[DEBUG] [1602480869.093960036]: XML file specifies lookup name (i.e. magic name) = rviz/MONO8.
[DEBUG] [1602480869.093969473]: XML file specifies lookup name (i.e. magic name) = rviz/RGBF32.
[DEBUG] [1602480869.093978492]: XML file specifies lookup name (i.e. magic name) = rviz/FlatColor.
[DEBUG] [1602480869.093987654]: XML file specifies lookup name (i.e. magic name) = rviz/AxisColor.
[DEBUG] [1602480869.093996352]: XML file specifies lookup name (i.e. magic name) = rviz/Temperature.
[DEBUG] [1602480869.094005599]: XML file specifies lookup name (i.e. magic name) = rviz/TF.
[DEBUG] [1602480869.094014349]: XML file specifies lookup name (i.e. magic name) = rviz/WrenchStamped.
[DEBUG] [1602480869.094025957]: XML file specifies lookup name (i.e. magic name) = rviz/PublishPoint.
[DEBUG] [1602480869.094033986]: XML file specifies lookup name (i.e. magic name) = rviz/FocusCamera.
[... 952 lines cut ...]
[DEBUG] [1602480869.293558709]: Attempting to create instance through low level multi-library class loader.
[DEBUG] [1602480869.293567507]: rviz/FlatColor maps to real class type rviz::FlatColorPCTransformer
[DEBUG] [1602480869.293589250]: Instance of type rviz::FlatColorPCTransformer created.
[DEBUG] [1602480869.293605703]: Attempting to create UNMANAGED instance for class rviz/Intensity.
[DEBUG] [1602480869.293617923]: Attempting to create instance through low level multi-library class loader.
[DEBUG] [1602480869.293626633]: rviz/Intensity maps to real class type rviz::IntensityPCTransformer
[DEBUG] [1602480869.293647411]: Instance of type rviz::IntensityPCTransformer created.
[DEBUG] [1602480869.293674807]: Attempting to create UNMANAGED instance for class rviz/MONO8.
[DEBUG] [1602480869.293686579]: Attempting to create instance through low level multi-library class loader.
[DEBUG] [1602480869.293695359]: rviz/MONO8 maps to real class type rviz::MONO8PCTransformer
[DEBUG] [1602480869.293716897]: Instance of type rviz::MONO8PCTransformer created.
[DEBUG] [1602480869.293728271]: Attempting to create UNMANAGED instance for class rviz/RGB8.
[DEBUG] [1602480869.293739103]: Attempting to create instance through low level multi-library class loader.
[DEBUG] [1602480869.293747845]: rviz/RGB8 maps to real class type rviz::RGB8PCTransformer
[DEBUG] [1602480869.293768816]: Instance of type rviz::RGB8PCTransformer created.
[DEBUG] [1602480869.293779781]: Attempting to create UNMANAGED instance for class rviz/RGBF32.
[DEBUG] [1602480869.293790525]: Attempting to create instance through low level multi-library class loader.
[DEBUG] [1602480869.293799271]: rviz/RGBF32 maps to real class type rviz::RGBF32PCTransformer
[DEBUG] [1602480869.293819941]: Instance of type rviz::RGBF32PCTransformer created.
[DEBUG] [1602480869.293830941]: Attempting to create UNMANAGED instance for class rviz/XYZ.
[DEBUG] [1602480869.293841780]: Attempting to create instance through low level multi-library class loader.
[DEBUG] [1602480869.293850426]: rviz/XYZ maps to real class type rviz::XYZPCTransformer
[DEBUG] [1602480869.293870725]: Instance of type rviz::XYZPCTransformer created.
[DEBUG] [1602480869.293949198]: MessageFilter [target=world ]: Cleared
```

rcywongaa, 2020-10-12:
This only happens when starting rviz with an existing rosmaster that previously had moveit running

rcywongaa, 2020-10-14:
Also happens when your rviz config has a moveit that was configured for a different arm than that currently running

rhaschke, 2020-10-19:
I cannot confirm this issue. The MotionPlanning plugin loads after a few seconds.

closed 2020-11-16 by rhaschke