ros-visualization/rviz #1549: rviz segmentation fault (core dumped) even though I already install OpenGL.

Han-Gyeol-621, 2020-10-14:
When I start rviz, I have a problem like below.

======================================================
gksruf621@DESKTOP-0IFPIFS:~$ rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-gksruf621'
[ INFO] [1602655014.398191700]: rviz version 1.13.13
[ INFO] [1602655014.398786400]: compiled against Qt version 5.9.5
[ INFO] [1602655014.399377900]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1602655014.416456200]: Forcing OpenGl version 0.
**Segmentation fault (core dumped)**
[image]

=======================================================

so I try to install OpenGL  since some people say it will help.

=================================================
gksruf621@DESKTOP-0IFPIFS:~$ glxinfo | grep "OpenGL version"
OpenGL version string: 1.4 (4.5.0 - Build 23.20.16.4849)
[image]

==============================================

but I still have a problem

this is my log.
===============================================
gksruf621@DESKTOP-0IFPIFS:~$ rviz -l
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-gksruf621'
[ INFO] [1602655021.597445800]: rviz version 1.13.13
[ INFO] [1602655021.597944000]: compiled against Qt version 5.9.5
[ INFO] [1602655021.598379000]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1602655021.613516300]: Forcing OpenGl version 0.
[ INFO] [1602655022.493480100]: Creating resource group General
[ INFO] [1602655022.494206800]: Creating resource group Internal
[ INFO] [1602655022.494714700]: Creating resource group Autodetect
[ INFO] [1602655022.495849800]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[ INFO] [1602655022.496775900]: Registering ResourceManager for type Material
[ INFO] [1602655022.497585000]: Registering ResourceManager for type Mesh
[ INFO] [1602655022.498075000]: Registering ResourceManager for type Skeleton
[ INFO] [1602655022.498771500]: MovableObjectFactory for type 'ParticleSystem' registered.
[ INFO] [1602655022.499435900]: ArchiveFactory for archive type FileSystem registered.
[ INFO] [1602655022.499898600]: ArchiveFactory for archive type Zip registered.
[ INFO] [1602655022.500351200]: ArchiveFactory for archive type EmbeddedZip registered.
[ INFO] [1602655022.500970900]: DDS codec registering
[ INFO] [1602655022.501381200]: FreeImage version: 3.17.0
[ INFO] [1602655022.501775000]: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
[ INFO] [1602655022.502905000]: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
[ INFO] [1602655022.504114800]: PVRTC codec registering
[ INFO] [1602655022.504526700]: ETC codec registering
[ INFO] [1602655022.504931000]: Registering ResourceManager for type HighLevelGpuProgram
[ INFO] [1602655022.505448600]: Registering ResourceManager for type Compositor
[ INFO] [1602655022.506565100]: MovableObjectFactory for type 'Entity' registered.
[ INFO] [1602655022.507220300]: MovableObjectFactory for type 'Light' registered.
[ INFO] [1602655022.507806300]: MovableObjectFactory for type 'BillboardSet' registered.
[ INFO] [1602655022.508418000]: MovableObjectFactory for type 'ManualObject' registered.
[ INFO] [1602655022.508963300]: MovableObjectFactory for type 'BillboardChain' registered.
[ INFO] [1602655022.509442800]: MovableObjectFactory for type 'RibbonTrail' registered.
[ INFO] [1602655022.510473200]: *-*-* OGRE Initialising
[ INFO] [1602655022.511099300]: *-*-* Version 1.9.0 (Ghadamon)
[ INFO] [1602655022.511813000]: OverlayElementFactory for type Panel registered.
[ INFO] [1602655022.512308200]: OverlayElementFactory for type BorderPanel registered.
[ INFO] [1602655022.512714400]: OverlayElementFactory for type TextArea registered.
[ INFO] [1602655022.513201600]: Registering ResourceManager for type Font
[ INFO] [1602655022.513629100]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL
[ INFO] [1602655022.521029200]: Installing plugin: GL RenderSystem
[ INFO] [1602655022.521939100]: OpenGL Rendering Subsystem created.
[ INFO] [1602655022.548979400]: Plugin successfully installed
[ INFO] [1602655022.549893700]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/Plugin_OctreeSceneManager
[ INFO] [1602655022.553576800]: Installing plugin: Octree Scene Manager
[... 27 lines cut ...]
[ INFO] [1602655022.576989100]:  *    3DNOW: no
[ INFO] [1602655022.577530600]:  * 3DNOWEXT: no
[ INFO] [1602655022.578021400]:  *     CMOV: yes
[ INFO] [1602655022.578626200]:  *      TSC: yes
[ INFO] [1602655022.579174200]:  *      FPU: yes
[ INFO] [1602655022.579697300]:  *      PRO: yes
[ INFO] [1602655022.580321400]:  *       HT: no
[ INFO] [1602655022.580889800]: -------------------------
[ INFO] [1602655022.581419100]: ******************************
*** Starting GLX Subsystem ***
******************************
[ INFO] [1602655022.582851000]: Registering ResourceManager for type Texture
[ INFO] [1602655022.583778600]: GLRenderSystem::_createRenderWindow "OgreWindow(0)", 1x1 windowed  miscParams: FSAA=4 contentScalingFactor=1.000000 externalGLControl= externalWindowHandle=3670017 parentWindowHandle=3670017
[ INFO] [1602655022.715052200]: GLXWindow::create used FBConfigID = 101
[ INFO] [1602655022.716025600]: GL_VERSION = 1.4 (4.5.0 - Build 23.20.16.4849)
[ INFO] [1602655022.717146900]: GL_VENDOR = Intel
[ INFO] [1602655022.717822400]: GL_RENDERER = Intel(R) UHD Graphics 620
[ INFO] [1602655022.718353100]: GL_EXTENSIONS = GL_ARB_depth_texture GL_ARB_draw_buffers GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_shadow GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_filter_anisotropic GL_ARB_texture_mirrored_repeat GL_ARB_texture_non_power_of_two GL_ARB_texture_rectangle GL_ARB_transpose_matrix GL_ARB_vertex_program GL_ARB_window_pos GL_EXT_abgr GL_EXT_bgra GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_clip_volume_hint GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_object GL_EXT_multi_draw_arrays GL_EXT_packed_pixels GL_EXT_point_parameters GL_EXT_rescale_normal GL_EXT_secondary_color GL_EXT_separate_specular_color GL_EXT_shadow_funcs GL_EXT_stencil_two_side GL_EXT_stencil_wrap GL_EXT_texture3D GL_EXT_texture_compression_s3tc GL_EXT_texture_edge_clamp GL_EXT_texture_env_add GL_EXT_texture_env_combine GL_EXT_texture_filter_anisotropic GL_EXT_texture_lod_bias GL_EXT_texture_rectangle GL_3DFX_texture_compression_FXT1 GL_ATI_draw_buffers GL_IBM_texture_mirrored_repeat GL_INGR_blend_func_separate GL_NV_blend_square GL_NV_texgen_reflection GL_NV_texture_rectangle GL_SGIS_generate_mipmap GL_SGIS_texture_border_clamp GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod GL_SUN_multi_draw_arrays
[ INFO] [1602655022.720583400]: Supported GLX extensions: GLX_ARB_create_context GLX_ARB_create_context_profile GLX_ARB_create_context_robustness GLX_ARB_fbconfig_float GLX_ARB_framebuffer_sRGB GLX_ARB_get_proc_address GLX_ARB_multisample GLX_EXT_create_context_es2_profile GLX_EXT_fbconfig_packed_float GLX_EXT_framebuffer_sRGB GLX_EXT_import_context GLX_EXT_visual_info GLX_EXT_visual_rating GLX_MESA_copy_sub_buffer GLX_OML_swap_method GLX_SGIS_multisample GLX_SGIX_fbconfig GLX_SGIX_pbuffer GLX_SGIX_visual_select_group GLX_SGI_make_current_read GLX_SGI_swap_control
[ INFO] [1602655022.721664100]: ***************************[ INFO] [1602655022.722009700]: *** GL Renderer Started ***[ INFO] [1602655022.722418300]: ***************************Segmentation fault (core dumped)
==============================================

* OS Version:  Ubuntu 18.04
* ROS Distro :  Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz : rviz version 1.13.13

Han-Gyeol-621, 2020-10-14:
Sorry this is my first time to ask a question github so I don't know why the log is big bold.

rhaschke, 2020-10-18:
Could you try to run rviz like this: `rviz --opengl 210`? This enforces an older OpenGL version to be used. We have seen issues in the past with Mesa OpenGL.

Ash-S, 2020-10-21:
Hi @rhaschke , I am facing the similar issue while tried to run 'rosrun rviz rviz' on my SBC.
Here are the details when i run 'rosrun rviz rviz'
[image]
Here are the details when i run gdb.
[image]
ooking forward to your kind suggestions to resolve this. Thanks

Han-Gyeol-621, 2020-10-29:
> Could you try to run rviz like this: `rviz --opengl 210`? This enforces an older OpenGL version to be used. We have seen issues in the past with Mesa OpenGL.

Thank you for your reply but it doesn't work

theg4sh, 2020-11-03:
@Han-Gyeol-621 need more details for that kind of tikets. Attach output of `gdb -ex 'run' -ex 'thread apply all bt' -ex 'q' rviz` to the ticket, please.

rhaschke, 2021-02-05:
Closing due to inactivity. Please feel free to reopen when you have more infos to reproduce.

closed 2021-02-05 by rhaschke

Fetiu, 2022-04-12:
Hi, I think I am experiencing exact the same issue.
I am running docker on macOS with XQuartz activated.

Seeing that `Forcing OpenGl version 0` and `GL Renderer Started` happened right before the segfault, it should be the same...although I do not have `(core dumped)` at the end, but I presume this might be a version or platform issue.

This is the log I get from `gdb -ex 'run' -ex 'thread apply all bt' -ex 'q' rviz` as @theg4sh requested.

[image]

This is the full version of logs
```
iros@docker-desktop:/$ gdb -ex 'run' -ex 'thread apply all bt' -ex 'q' rviz
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rviz...
(No debugging symbols found in rviz)
Starting program: /opt/ros/noetic/bin/rviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffee67a700 (LWP 228)]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-iros'
[ INFO] [1649752366.330507279]: rviz version 1.14.14
[ INFO] [1649752366.330708371]: compiled against Qt version 5.12.8
[ INFO] [1649752366.330756380]: compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffed663700 (LWP 230)]
[New Thread 0x7fffece62700 (LWP 231)]
[New Thread 0x7fffdffff700 (LWP 232)]
[New Thread 0x7fffdf7fe700 (LWP 233)]
[ INFO] [1649752366.368212371]: Forcing OpenGl version 0.
[New Thread 0x7fffdeffd700 (LWP 234)]
[Detaching after fork from child process 235]
--Type <RET> for more, q to quit, c to continue without paging--

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()

Thread 7 (Thread 0x7fffdeffd700 (LWP 234)):
#0  0x00007ffff6e589cf in __GI___poll (fds=0x7fffd40029e0, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007ffff460b36e in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff460b4a3 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7413435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff73ba3ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff71f2785 in QThread::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007fffeeff5efa in ?? () from /lib/x86_64-linux-gnu/libQt5DBus.so.5
#7  0x00007ffff71f39d2 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff68a4609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#9  0x00007ffff6e65163 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

--Type <RET> for more, q to quit, c to continue without paging--
Thread 6 (Thread 0x7fffdf7fe700 (LWP 233)):
#0  futex_abstimed_wait_cancelable (private=<optimized out>, abstime=0x7fffdf7fda60, clockid=<optimized out>, expected=0, futex_word=0x55555561bdc8) at ../sysdeps/nptl/futex-internal.h:320
#1  __pthread_cond_wait_common (abstime=0x7fffdf7fda60, clockid=<optimized out>, mutex=0x55555561bd78, cond=0x55555561bda0) at pthread_cond_wait.c:520
#2  __pthread_cond_timedwait (cond=0x55555561bda0, mutex=0x55555561bd78, abstime=0x7fffdf7fda60) at pthread_cond_wait.c:656
#3  0x00007ffff69c8baf in ros::CallbackQueue::callAvailable(ros::WallDuration) () from /opt/ros/noetic/lib/libroscpp.so
#4  0x00007ffff6a03fbf in ros::internalCallbackQueueThreadFunc() () from /opt/ros/noetic/lib/libroscpp.so
#5  0x00007ffff6c7c43b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#6  0x00007ffff68a4609 in start_thread (arg=<optimized out>) at pthread_create.c:477
[... 35 lines cut ...]
#5  0x00007ffff68a4609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007ffff6e65163 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 1 (Thread 0x7fffef2a1a80 (LWP 224)):
#0  0x0000000000000000 in ?? ()
#1  0x00007fffddc869d3 in ?? () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#2  0x00007fffddc60e9c in Ogre::GLStateCacheManager::switchContext(long) () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#3  0x00007fffddc540e0 in Ogre::GLRenderSystem::initialiseContext(Ogre::RenderWindow*) () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#4  0x00007fffddc5ca1b in Ogre::GLRenderSystem::_createRenderWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::all [... cut ...]
#10 0x00007ffff7ec6916 in rviz::QtOgreRenderWindow::QtOgreRenderWindow(QWidget*) () from /opt/ros/noetic/lib/librviz.so
#11 0x00007ffff7eebf1f in rviz::RenderPanel::RenderPanel(QWidget*) () from /opt/ros/noetic/lib/librviz.so
#12 0x00007ffff7f38fc3 in rviz::VisualizationFrame::initialize(QString const&) () from /opt/ros/noetic/lib/librviz.so
#13 0x00007ffff7f42e23 in rviz::VisualizerApp::init(int, char**) () from /opt/ros/noetic/lib/librviz.so
#14 0x00005555555585cb in main ()
A debugging session is active.
```

Hope I can get some responses.
Thanks in advance.

ali-mubarak001, 2025-07-24:
Hey, I got the same issue:

```
root@dc8e4213570a:/opt# export LIBGL_ALWAYS_SOFTWARE=1
root@dc8e4213570a:/opt# export QT_XCB_GL_INTEGRATION=none
root@dc8e4213570a:/opt# export QT_OPENGL=software
root@dc8e4213570a:/opt# export QT_X11_NO_MITSHM=1
root@dc8e4213570a:/opt# rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [1753383171.119786169]: rviz version 1.14.26
[INFO] [1753383171.120251294]: compiled against Qt version 5.12.8
[INFO] [1753383171.120307377]: compiled against OGRE version 1.9.0 (Ghadamon)
[INFO] [1753383171.141887169]: Forcing OpenGl version 0.
Segmentation fault
root@dc8e4213570a:/opt# ls
```