ros-visualization/rviz #1550: rviz frame rate  drops rapidly and display slowly

jacklidar, 2020-10-15:
Hi，We use rviz with ubuntu 16.04 + virtualbox to run rviz by command 'roslaunch loam_horizon loam_livox_horizon.launch', rviz started displaying at 30fps, and then start dropped slowly to 1fps and display slowly. If you click reset, the frame rate will be restored to 30fps. We use rosbag to run with 'rosbag play 2020_parking_lot.bag', the issue also reproduce too. 

How can I increase the frame rate of the display? Is there any parameter settings for rviz? How can we resolve this issue?

Thanks

rhaschke, 2020-10-15:
The core problem is that you are trying to run rviz from a VM. Probably, the VM cannot grant direct access to the GPU, making any rendering slow. The more data points need to be visualized (I guess your bag publishes a point cloud?) the slower it gets.

closed 2020-10-15 by rhaschke

jacklidar, 2020-10-15:
Yes, the bag publishes is a point cloud. Thanks