ros-visualization/rviz #1552: Drag and drop of panels does not allow side-by-side positioning

paland3, 2020-10-20:
Hello, I have two custom panels in rviz (one camera and one image) which I would like to show side by side. I can easily rearrange them in a top bottom fashion or in separate tabs, but cannot make them in to a side by side layout. 

I can do this:
[image]

**But I would like to have the two "camera" panel next to each other, horizontally. Is there a way to do this?**

### Your environment
* Ubuntu 18.04
* ROS Distro:  Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

```
[ INFO] [1603181482.756307440]: rviz version 1.13.13
[ INFO] [1603181482.756420956]: compiled against Qt version 5.9.5
[ INFO] [1603181482.756462844]: compiled against OGRE version 1.9.0 (Ghadamon)
```

rhaschke, 2020-10-20:
This is a limitation of Qt's DockWidgets: https://doc.qt.io/qt-5/qdockwidget.html#details.
On the left and right they can be stacked vertically, on the bottom and top, they can be stacked horizontally:

[image]

closed 2020-10-20 by rhaschke

paland3, 2020-10-20:
@rhaschke  Thanks, this works almost perfectly! (stacking them at the bottom). 

Is there a way to hide the main 3d window? I.e. to close/remove it all the way. See attached picture: cannot remove the bottom bar (3d window minimised as much as I can).
[image]

When an image and the 3d window is stacked horizontally, I can minimise the 3d window completely:
[image]

I do not need the 3d view in this setup. Thanks again!

rhaschke, 2020-10-20:
No, there is no way to hide the central widget. You could put one camera in the left dock and one in the right and then "hide" the central widget in between:
[image]

paland3, 2020-10-20:
Nice, thanks again!