ros-visualization/rviz #1553: 'rosrun rviz rviz' Segmentation Fault (Ubuntu Bionic, iMX6 SBC, ROS Melodic)

Ash-S, 2020-10-21:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: e.g. Ubuntu 18.04 Bionic
* ROS Distro: [Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
``` **command run: rosrun rviz rviz**
[ INFO] [1603250787.420411222]: rviz version 1.13.13  
[ INFO] [1603250787.420690906]: compiled against Qt version 5.9.5 
[ INFO] [1603250787.420791913]: compiled against OGRE version 1.9.0 (Ghadamon) 
[ INFO] [1603250787.514283713]: Forcing OpenGl version 0.
[ INFO] [1603250788.270683306]: Stereo is NOT SUPPORTED 
[ INFO] [1603250788.272516420]: OpenGl version: 3.1 (GLSL 1.4).                                                 │
**Segmentation fault** 

**command run: gdb /opt/ros/melodic/lib/rviz/rviz**
GNU gdb (Ubuntu 8.1-0ubuntu3.2) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/melodic/lib/rviz/rviz...(no debugging symbols found)...done.

**(gdb) r**
Starting program: /opt/ros/melodic/lib/rviz/rviz
Program received signal SIGSEGV, Segmentation fault.
0x76fd8f4e in ?? () from /lib/ld-linux-armhf.so.3
**(gdb) bt**
#0  0x76fd8f4e in ?? () from /lib/ld-linux-armhf.so.3
#1  0x76fd8ef8 in ?? () from /lib/ld-linux-armhf.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
    ```
* If source build, which git commit?
followed this to install ROS melodic on my SBC 
http://wiki.ros.org/melodic/Installation/Ubuntu

rhaschke, 2020-10-21:
To get a helpful backtrace, you should compile rviz from source, with debugging symbols.
Is it really useful to run rviz on arm? I guess it doesn't provide any GPU support, does it?

Ash-S, 2020-10-21:
Hi @rhaschke , thanks for your reply.
well, Could you please let me know some steps to compile rviz from source? and yes, since I am developing ROS based application on arm SBC, i need to run rviz on arm. The GPU used in the processor is from Vivante.

rhaschke, 2020-10-21:
I suggest reading the basic tutorial: http://wiki.ros.org/catkin/Tutorials/using_a_workspace

Tobias-Fischer, 2020-12-06:
Hi @Ash-S, you might want to check out https://github.com/RoboStack/ros-noetic where we build ROS packages based on conda-forge. Drop by https://gitter.im/RoboStack/ if you have questions / want to contribute. Disclaimer: So far we don't have all dependencies for rviz ready, so we don't provide rviz just yet on ARM.

rhaschke, 2021-02-05:
Closing due to inactivity. Please feel free to reopen and provide more detailed infos.

closed 2021-02-05 by rhaschke

Tobias-Fischer, 2021-02-06:
@Ash-S we now have rviz for ARM precompiled. Let me know if it works for you.