ros-visualization/rviz #1554: How to solve the conflict between rviz rendering and qt interface rendering.

lrm2017, 2020-10-24:
Describe your issue here and explain how to reproduce it.

When I integrate rviz in the qt interface and initialize rviz, loading qml resources in another interface, the following phenomenon occurs, while rviz becomes stuck and does not work.
[image]

The interface in question is as follows.
[image]

The normal interface is as follows.
[image]

Based on rviz's development, I am a beginner.

### Your environment
* OS Version: e.g. Ubuntu 16.04
* ROS Distro: Kinetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
   Qt 5.9.9
    paste your console output here
    https://github.com/lrm2017/-/tree/master/catkin_ws/src/Ros_Qt5_Gui
* If source build, which git commit?

rhaschke, 2020-10-24:
This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here.
If you re-post over there, please provide more details. How do integrate? Based on C++ or python? Which functions / classes do you use for integration?

closed 2020-10-24 by rhaschke

lrm2017, 2020-10-24:
> This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here.
> If you re-post over there, please provide more details. How do integrate? Based on C++ or python? Which functions / classes do you use for integration?

I'm so sorry, but the first time I asked a question, I didn't know the rules and wasted your time. Thank you, I will try to provide more detailed questions.