ros-visualization/rviz #1557: window blinking while move out of the rviz window

lianghongzhuo, 2020-11-15:
Describe your issue here and explain how to reproduce it.

### Your environment
* Ubuntu 18.04
* ROS Distro: Melodic
* RViz 1.13.14, Qt, OGRE 1.9.0
  
When moving the mouse into rviz will display some help info as seen in the picture below:
[image]
This is helpful, however the content `Left-Click:Rotate. Middle click: ....` is not at the same line as the `Reset` button on the left.
This will leads the screen blinking when moving the mouse out of the rviz window:
[image]
As seen in the above picture, the help info disappears, which leads to a screen blinking.
I do not know whether it's only happening on my computer or a general issue.

rhaschke, 2020-11-15:
I cannot reproduce this issue as well. What exactly is blinking? The text or the window (as stated in title)?
What does it mean for the window to blink? Does the full window disappear and appear?

lianghongzhuo, 2020-11-15:
[screencast_00000.zip](https://github.com/ros-visualization/rviz/files/5543002/screencast_00000.zip)
Please see the screen shot video.

rhaschke, 2020-11-16:
Indeed, I think you have chosen a too large font for the text. This increases the widget size causing the flickering you observe.
Closing here.

closed 2020-11-16 by rhaschke

lianghongzhuo, 2020-11-16:
It's nothing to do with the font size, but the issuse with the font type. When I chagne the font to the ubuntu default font "ubuntu" the content now can display at the same line as the `reset` buttom. And then no more blinking of the screen.