ros-visualization/rviz #1561: Stack Smashing detected, when starting rviz

BestVanRome, 2020-11-17:
Hello guys and gals, 
the following output , when I' trying to start rviz:
```
[ INFO] [1605605647.233095655]: rviz version 1.14.1
[ INFO] [1605605647.233166559]: compiled against Qt version 5.12.8
[ INFO] [1605605647.233188680]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1605605647.244357543]: Forcing OpenGl version 0.
*** stack smashing detected ***: terminated
Abgebrochen (Speicherabzug geschrieben)
```
rosdistro: rosnoetic
rosversion: 1.15.8
OS Version: Ubuntu 20.04 LTS

When I run "roswtf", I get the following output:

```
Loaded plugin tf.tfwtf
No package or stack in the current directory
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
ERROR: connection refused to [http://agas-ThinkPad-T540p:33347/]
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING No tf messages

Found 3 error(s).

ERROR Could not contact the following nodes:
 * /rviz_1605606036063192314

ERROR The following nodes should be connected but aren't:
 * /rviz_1605606036063192314->/rosout (/rosout)

ERROR Errors connecting to the following services:
 * service [/rviz_1605606036063192314/get_loggers] appears to be malfunctioning: Unable to communicate with service [/rviz_1605606036063192314/get_loggers], address [rosrpc://agas-ThinkPad-T540p:48697]
 * service [/rviz_1605606036063192314/set_logger_level] appears to be malfunctioning: Unable to communicate with service [/rviz_1605606036063192314/set_logger_level], address [rosrpc://agas-ThinkPad-T540p:48697]
```
Thanks in advance for your support!

rhaschke, 2020-11-17:
This output is not yet very helpful. But, as it dies before connecting to the RenderSystem, I suspect an OpenGL issue.
Which system you are running on, i.e. which Linux, which graphics card, which OpenGL driver (`glxinfo -B`)?
Also, try `rviz --opengl 210`.

BestVanRome, 2020-11-17:
See obove ;-) :
> Ubuntu 20.04 LTS   

and to be exact: **Ubuntu 20.04.1 LTS focal-release**  

I'll add the **kernel** :
5.8.12-050812-generic

**Graphics:**
```
00:02.0 VGA compatible controller: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
01:00.0 VGA compatible controller: NVIDIA Corporation GK208M [GeForce GT 730M] (rev a1)
```

`glxinfo -B` returns:  
```
name of display: :0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  41
  Current serial number in output stream:  42
```

Does this   mean a problem with the Graphics? 

`rviz --opengl 210` does not change anything.

Here a screenshot from my drivers of the graphics:
[image]

Hope this helps.

rhaschke, 2020-11-17:
Yep, glxinfo failing means that your OpenGL config is broken. Please fix this first.

BestVanRome, 2020-11-17:
It's strange, I changed nothing, just rebooted a few times.
Now glxinfo works ... very strange.
Here's the output of `glxinfo -B`:
```
ame of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) HD Graphics 4600 (HSW GT2) (0x416)
    Version: 20.0.8
    Accelerated: yes
    Video memory: 1536MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 4600 (HSW GT2)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 3.0 Mesa 20.0.8
OpenGL shading language version string: 1.30
OpenGL context flags: (none)

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 20.0.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10

```

BestVanRome, 2020-11-17:
Changed to dedicated graphics NVIDIA GT 730M but still no change.
`roswtf` tells one of these messages: 
``
service [/rviz_1605627927787921964/get_loggers] appears to be malfunctioning: Unable to communicate with service
``
Does it give a clue of how to proceed?

BestVanRome, 2020-11-17:
Just created a core file.

[core.zip](https://github.com/ros-visualization/rviz/files/5554912/core.zip)

rhaschke, 2020-11-18:
I have noticed that you are using an outdated version of rviz. Please, do a full system update first, i.e.
sudo apt update
sudo apt upgrade

BestVanRome, 2020-11-18:
> I have noticed that you are using an outdated version of rviz. Please, do a full system update first, i.e.
> sudo apt update
> sudo apt upgrade

No improvement :-(

rhaschke, 2020-11-18:
As we haven't seen such an issue before, I'm afraid that this is related to your system, which is probably somehow screwed.
Have you tried rviz on another system, or did you try a clean install from scratch already?

BestVanRome, 2020-11-18:
Yes, I tried to reinstall the whole ROS-distro from scratch.
Unfortunately, I don't have a second system on my Laptop.
I have to get it running, that's why I rely on your support.

In my search, I stumbled over an old commit of allenh1/rviz_stack_smashing (https://gist.github.com/allenh1/08aecdf09bcfe0154573e8f7ad83cb40), where he uses "valgrind" to narrow the cause.

So I tried it aswell and here is what I got in short-version:
```
==10147== Memcheck, a memory error detector
==10147== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==10147== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==10147== Command: rviz
==10147== 
--10147-- WARNING: unhandled amd64-linux syscall: 315
--10147-- You may be able to write your own handler.
--10147-- Read the file README_MISSING_SYSCALL_OR_IOCTL.
--10147-- Nevertheless we consider this a bug.  Please report
--10147-- it at http://valgrind.org/support/bug_reports.html.
[ INFO] [1605691709.834479097]: rviz version 1.14.1
[ INFO] [1605691709.849809489]: compiled against Qt version 5.12.8
[ INFO] [1605691709.850929854]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1605691710.283206841]: Forcing OpenGl version 0.
*** stack smashing detected ***: terminated
==10147== 
==10147== Process terminating with default action of signal 6 (SIGABRT)
==10147==    at 0x592818B: raise (raise.c:51)
==10147==    by 0x5907858: abort (abort.c:79)
==10147==    by 0x59723ED: __libc_message (libc_fatal.c:155)
==10147==    by 0x5A14B49: __fortify_fail (fortify_fail.c:26)
==10147==    by 0x5A14B15: __stack_chk_fail (stack_chk_fail.c:24)
==10147==    by 0x4A43673: rviz::YamlConfigReader::readStream(rviz::Config&, std::istream&, QString const&) (in /opt/ros/noetic/lib/librviz.so)
==10147==    by 0x4A43866: rviz::YamlConfigReader::readFile(rviz::Config&, QString const&) (in /opt/ros/noetic/lib/librviz.so)
==10147==    by 0x4A2979D: rviz::VisualizationFrame::loadPersistentSettings() (in /opt/ros/noetic/lib/librviz.so)
==10147==    by 0x4A2D8EB: rviz::VisualizationFrame::initialize(QString const&) (in /opt/ros/noetic/lib/librviz.so)
==10147==    by 0x4A37764: rviz::VisualizerApp::init(int, char**) (in /opt/ros/noetic/lib/librviz.so)
==10147==    by 0x10C5AA: main (in /opt/ros/noetic/bin/rviz)
==10147== 
==10147== HEAP SUMMARY:
==10147==     in use at exit: 7,637,412 bytes in 41,175 blocks
==10147==   total heap usage: 711,729 allocs, 670,554 frees, 56,967,469 bytes allocated
==10147== 
==10147== LEAK SUMMARY:
==10147==    definitely lost: 256 bytes in 1 blocks
==10147==    indirectly lost: 32 bytes in 1 blocks
==10147==      possibly lost: 7,152 bytes in 41 blocks
==10147==    still reachable: 7,547,340 bytes in 40,458 blocks
==10147==                       of which reachable via heuristic:
==10147==                         length64           : 4,960 bytes in 82 blocks
==10147==                         newarray           : 2,096 bytes in 51 blocks
==10147==         suppressed: 0 bytes in 0 blocks
==10147== Rerun with --leak-check=full to see details of leaked memory
==10147== 
==10147== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Abgebrochen (Speicherabzug geschrieben)
```
I also did a rerun of valgrind with full-leak-check. 
It's stored here: 
[rviz_dump_leak_check_full.txt](https://github.com/ros-visualization/rviz/files/5559138/rviz_dump_leak_check_full.txt)

Maybe you can give me at least a hint, what I have to check on my side, because right now I don't have a clue...

BestVanRome, 2020-11-25:
Ok, seems like I have a messed up system-config. 
I set up a new environment in docker now. (Ubuntu 18.04 + ros-melodic). 
It works from there.

closed 2020-11-25 by BestVanRome