ros-visualization/rviz #1562: Segmentation fault when using LINE_LIST as an interactive marker

Sean-Katagiri, 2020-11-18:
I've been attempting to visualize a graph using interactive markers, and have noticed issues specifically with using the LINE_LIST marker type as an interactive marker. When a LINE_LIST interactive marker is added to the rviz scene the following warning appears in the rviz terminal: 
```
[ WARN] [1605635701.622176693]: Marker 'edge_0/5': scale.y and scale.z of LINE_LIST or LINE_STRIP are ignored.
```
The issue is, I only ever set the "scale.x" of the marker and never touch "scale.y" or "scale.z". The warning poses no issue when it is a single edge that is added to the scene, but after 50+ edges are added **rviz crashes with a segfault**. 

I've also tried setting "scale.y" and "scale.z" to 0.0 manually (and confirmed that those values were in fact 0.0 by checking "/update_full" topic), but this unfortunately made no change to the outcome and the warning messages still appear. 

I'm happy to simply use Marker type CYLINDER in place of LINE_STRIP/LINE_LIST, however I wondered if this was a bug and whether it was worth bringing up here. 

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1605635700.243055836]: rviz version 1.13.14
[ INFO] [1605635700.243103163]: compiled against Qt version 5.9.5
[ INFO] [1605635700.243115506]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1605635700.247199098]: Forcing OpenGl version 0.
[ INFO] [1605635700.764818782]: Stereo is NOT SUPPORTED
[ INFO] [1605635700.764998311]: OpenGl version: 4.6 (GLSL 4.6).
```

rhaschke, 2020-11-18:
Thanks for reporting this bug. Could you please provide a bag or yaml file to reproduce this issue.
Is this only related to interactive markers or to markers in general?

Sean-Katagiri, 2020-11-18:
Thank you for the swift reply. On further inspection the segfault may be caused by poor code design on my part so I need to do more testing after I make the necessary changes.

The rviz warning behavior can be seen using this short rosbag however.
* Frame: /base_link
* Interactive marker topic: /pose_graph_visualizer/update

[2020-11-18-12-02-04.zip](https://github.com/ros-visualization/rviz/files/5559840/2020-11-18-12-02-04.zip)

rhaschke, 2020-12-17:
The warning about non-zero scales is a bad interaction between #1275 and the `interactive_markers` package auto-completing those zero scales to 1.0... I will remove the corresponding warnings from rviz.
Regarding the crash: I wasn't able to reproduce it. Please provide a backtrace and also try running rviz with option `--opengl 310`. This issue might be related to #1572.

rhaschke, 2021-02-05:
Closing due to inactivity. If you can still reproduce the segfault, please provide more info.

closed 2021-02-05 by rhaschke