ros-visualization/rviz #1563: Markers not rendered properly on ROS Noetic

VladyslavUsenko, 2020-11-22:
Markers (CUBE_LIST, SPHERE_LIST) not rendered properly with ROS Noetic. Instead of actual primitives, they are rendered as white dots. The same markers work fine with ROS Melodic. See the attached screenshots (top - melodic, bottom noetic):

[image]
[image]

The bag and rviz files for reproducing the issue are attached below: 

https://drive.google.com/file/d/1us-gtL9U1jAzFcq6WAl4wWNgKX6QEEPg/view?usp=sharing

[simulation.rviz.zip](https://github.com/ros-visualization/rviz/files/5579822/simulation.rviz.zip)

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1606062535.201571365]: rviz version 1.14.1
[ INFO] [1606062535.201613247]: compiled against Qt version 5.12.8
[ INFO] [1606062535.201625189]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1606062535.209114479]: Forcing OpenGl version 0.
[ INFO] [1606062535.382282171]: Stereo is NOT SUPPORTED
[ INFO] [1606062535.382330846]: OpenGl version: 4.6 (GLSL 4.6).
```

rhaschke, 2020-11-22:
This looks like a duplicate of #1508. Please check the solution there: running rviz with option `--opengl 210`.

VladyslavUsenko, 2020-11-22:
Yes, `--opengl 210` fixes the issue. Thanks!

closed 2020-11-22 by VladyslavUsenko