ros-visualization/rviz #1564: Ubuntu 18.04: rosrun rviz rviz  => RViz: waitiing for master (Error)

Tricia279, 2020-11-23:
Try to run:
rosrun rviz rviz 

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libEGL warning: DRI2: failed to authenticate
[ INFO] [1606135744.824416237]: rviz version 1.13.14
[ INFO] [1606135744.824817744]: compiled against Qt version 5.9.5
[ INFO] [1606135744.824941753]: compiled against OGRE version 1.9.0 (Ghadamon)
root@robert-desktop:/# rosrun rviz system_info.sh
[rosrun] Couldn't find executable named system_info.sh below /opt/ros/melodic/share/rviz
root@robert-desktop:/# rosrun rviz rviz -l
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libEGL warning: DRI2: failed to authenticate
[ INFO] [1606136415.831652269]: rviz version 1.13.14
[ INFO] [1606136415.832007630]: compiled against Qt version 5.9.5
[ INFO] [1606136415.832127056]: compiled against OGRE version 1.9.0 (Ghadamon)
root@robert-desktop:/# ulimit -c unlimited
root@robert-desktop:/# rosrun rviz rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libEGL warning: DRI2: failed to authenticate
[ INFO] [1606136492.147837686]: rviz version 1.13.14
[ INFO] [1606136492.148065549]: compiled against Qt version 5.9.5
[ INFO] [1606136492.148129350]: compiled against OGRE version 1.9.0 (Ghadamon)

It appears a window with the text: "RViz: waitiing for master 
                                                        Could not contact ROS master [http://localhost: 11311], retrying..."

=> only aborting is possible

I can not say, if it is a bug or not.
Please try and find out...

### Your environment
* Ubuntu 18.04
* ROS Melodic
* RViz

rhaschke, 2020-11-23:
1. Why do you want to execute rviz as root? That's the reasone that `XDG_RUNTIME_DIR` is not defined.
2. Obviously you need to start the ROS master, i.e. run `roscore`, before starting any other ROS app.
3. Cancelling the window "RViz: waiting for master" works without problems for me, i.e. closes rviz immediately.

closed 2020-11-23 by rhaschke

Tricia279, 2020-11-24:
If I run roscore, nothing happens:

robert@robert-desktop:~$ roscore
... logging to /home/robert/.ros/log/9850c254-2e2b-11eb-947d-b827eb67c271/roslaunch-robert-desktop-2051.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robert-desktop:43199/
ros_comm version 1.14.10

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES

auto-starting new master
process[master]: started with pid [2204]
ROS_MASTER_URI=http://robert-desktop:11311/

setting /run_id to 9850c254-2e2b-11eb-947d-b827eb67c271
process[rosout-1]: started with pid [2216]
started core service [/rosout]

Is another ROS app missing ?

If I do rviz as a NOT-root-user the same thinh happens:

robert@robert-desktop:~$ rosrun rviz rviz
libEGL warning: DRI2: failed to authenticate
[ INFO] [1606205367.866298217]: rviz version 1.13.14
[ INFO] [1606205367.866742354]: compiled against Qt version 5.9.5
[ INFO] [1606205367.866867777]: compiled against OGRE version 1.9.0 (Ghadamon)

Any ideas ?

rhaschke, 2020-11-26:
Please validate your OpenGL setup. DRI failing points to a more fundamental issue in your system.

Tricia279, 2020-11-26:
Thank you for your comment, but I do not understand in which relation OpenGL stands to linux / Ubuntu 18.04 and DRI ?
And the most essential question is what to do then or next ? Deinstallation of something...

rhaschke, 2020-11-26:
DRI is short for Direct Rendering Interface. So it relates to OpenGL rendering. Obviously, rviz cannot open an OpenGL window, which is required for rendering. I cannot further help fixing your issue, because this is very system specific now. I suggest to start from a fresh installation and then verify the `glxinfo` and `glxgears` work as expected. Please google for more general help on OpenGL issue.