ros-visualization/rviz #1565: Mising rviz_export.h on jenkins pipeline.

maximilianwulf, 2020-11-23:
Our jenkins [pipeline is failing ](http://build.ros.org/job/Kdev__grid_map__ubuntu_xenial_amd64/44/console) because of the following issue. I cannot recreate it on my machine. I suspect the jenkin server is missing something, but I cannot figure out what. 

[image]

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Kinetic | Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    paste your console output here
    ```
* If source build, which git commit?

YoshuaNava, 2020-11-23:
Hi,

Max is the main maintainer of grid_map, a widely available ROS package. I also tried to help debug this issue, but I couldn't find a solution.

It seems that the header file is generated here: https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/default_plugin/CMakeLists.txt#L72

And we tried including it from this directory: https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/default_plugin/CMakeLists.txt#L90

But our attempts were not successful.

rhaschke, 2020-11-26:
`rviz` installs this file to `/opt/ros/<ros version>/include/rviz/rviz_export.h`. I don't have a Kinetic release lying around to confirm on Kinetic. But for Melodic that's the case. Looking at the build server, I noticed that the Melodic build [compiles fine](http://build.ros.org/job/Mdev__grid_map__ubuntu_bionic_amd64/). So, can you confirm that only Kinetic is affected?

We didn't release rviz into Kinetic since March 2019. So, I guess you changed something in your cmake config meanwhile.
Please check your config with [catkin_lint](https://fkie.github.io/catkin_lint).

maximilianwulf, 2020-11-30:
> So, can you confirm that only Kinetic is affected?

Yes, I can confirm this. 

> We didn't release rviz into Kinetic since March 2019. So, I guess you changed something in your cmake config meanwhile.
Please check your config with catkin_lint.

Will look into it. 

Thanks for the help.

rhaschke, 2021-02-05:
@maximilianwulf, did you resolve this issue? Can we close it?

maximilianwulf, 2021-02-08:
@rhaschke no we didn't solve it yet. We will tag a stable kinetic version. Thank you for your help.

closed 2021-02-08 by maximilianwulf