ros-visualization/rviz #1566: RViz doesn't interpolate

awied1404, 2020-12-05:
Hello guys,

I have a little problem. I am using the Kuka YouBot and want to calculate a trajectory.
I created a catkin package with MoveIt setup assistant and after setting a goal pose via RViz GUI and 
calculating the trajectory I look at the path and see the following:
[image]

In my opinion a interpolation made when executing this trajectory would detect this collision but somehow it doesn't. 

My environment
* Ubuntu 18.04
* ROS Distro: Melodic
* RViz, MoveIt

My fake_controllers.yaml file:
rate: 20
controller_list:
  - name: fake_youbot_arm_controller
    type: via points
    joints:
      - arm_joint_1
      - arm_joint_2
      - arm_joint_3
      - arm_joint_4
      - arm_joint_5
  - name: fake_gripper_controller
    joints:
      - gripper_finger_joint_l
      - gripper_finger_joint_r
initial:  # Define initial robot poses.
  - group: youbot_arm
    pose: home
  - group: gripper
    pose: home

I also tried the default type for the controllers (interpolate) but it doesn't work either.

rhaschke, 2020-12-05:
This is a MoveIt, not rviz related problem. Please post this issue over there. Doing so, please provide more details:
- Which planner did you use?
- Do you have warnings on the console where you started the `move_group` node?
As far as I remember, the fake controllers don't interpolate, but just display all via points of the planned trajectory.
However, the key issue here is that the planned trajectory has such a huge jump at all.

closed 2020-12-05 by rhaschke

awied1404, 2020-12-06:
Okay thank you for your help.
So RViz doesn't have the ability to interpolate at all?

rhaschke, 2020-12-06:
No rviz doesn't interpolate on its own. It's just displaying what it is commanded from elsewhere.