ros-visualization/rviz #1567: Unable to use load_config/save_config services

danem, 2020-12-07:
### Your environment
* OS Version: Ubuntu 18.04.1
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1607377547.846715532]: rviz version 1.13.15
    [ INFO] [1607377547.846743312]: compiled against Qt version 5.9.5
    [ INFO] [1607377547.846749297]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1607377547.852355452]: Forcing OpenGl version 0.
    [ INFO] [1607377548.473284250]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
    [ INFO] [1607377548.513391268]: Stereo is NOT SUPPORTED
    [ INFO] [1607377548.513471352]: OpenGl version: 4.6 (GLSL 4.6).
    ```
* Commit: 2fe6d33f1ca7ceccd916cea94c199ac8a2f25ca5

When building from source from commit 2fe6d33f1ca7ceccd916cea94c199ac8a2f25ca5, I cannot seem to use the `load_config` and `save_config` services provided by rviz. For instance, here is the output of rosservice:

    rosservice call /rviz_ttt_20736_1919514447758278227/load_config 
    ERROR: Unable to load type [rviz/SendFilePath].
    Have you typed 'make' in [rviz]?

Meaning it likely fails around [here](https://github.com/ros/ros_comm/blob/melodic-devel/tools/rosservice/src/rosservice/__init__.py#L373).

Additionally I cannot import the `rviz.srv` package from python scripts. I am sure that I've sourced my environment correctly, and I can see that the service header files and python files are placed within the `devel` folder of my catkin workspace:

    root@ttt:/workspace# ls devel/.private/rviz/lib/python2.7/dist-packages/rviz/
    __init__.py  librviz_sip.exp  librviz_sip.so  srv
    root@ttt:/workspace# ls devel/.private/rviz/lib/python2.7/dist-packages/rviz/srv/
    _SendFilePath.py  __init__.py
    root@ttt:/workspace# ls devel/.private/rviz/include/rviz/
    SendFilePath.h  SendFilePathRequest.h  SendFilePathResponse.h  default_plugin  rviz_export.h

Oddly enough, `rossrv` does seem to know about the service type though:

    root@ttt:/workspace# rossrv info rviz/SendFilePath
    std_msgs/String path
      string data
    ---
    bool success

rhaschke, 2020-12-08:
This all works with the binary installation, i.e. triggering the service and importing `rviz.srv` from python.
Please double check your ROS environment.

danem, 2020-12-08:
Any tips on what to look for? I haven't run into a similar problem despite making extensive use of ROS's service capabilities. 

I've also made sure that a previous binary installation of rviz wasn't conflicting with my source build.

`roswtf` doesn't report anything interesting either:

    root@ttt:/workspace# roswtf
    Loaded plugin tf.tfwtf
    No package or stack in context
    ================================================================================
    Static checks summary:

    No errors or warnings
    ================================================================================
    Beginning tests of your ROS graph. These may take awhile...
    analyzing graph...
    ... done analyzing graph
    running graph rules...
    ... done running graph rules
    running tf checks, this will take a second...
    ... tf checks complete

    Online checks summary:

    Found 1 warning(s).
    Warnings are things that may be just fine, but are sometimes at fault

    WARNING The following node subscriptions are unconnected:
       * /rviz_ttt_20736_1919514447758278227:
           * /drop_markers_array
           * /grasp_markers_array
           * /survey_markers_array
           * /armani_moveit_gui_prompt
           * /recognized_object_array
 
Finally for good measure, here is my catkin config:
```
root@ttt:/workspace# catkin config
---------------------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /workspace/devel/.private/catkin_tools_prebuild:/opt/ros/melodic
Workspace:                   /workspace
---------------------------------------------------------------------------------------------
Build Space:        [exists] /workspace/build
Devel Space:        [exists] /workspace/devel
Install Space:      [unused] /workspace/install
Log Space:          [exists] /workspace/logs
Source Space:       [exists] /workspace/src
DESTDIR:            [unused] None
---------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------------------------------
Workspace configuration appears valid.

```

rhaschke, 2020-12-08:
Give me a second. Initially I tried to reproduce on Noetic. Not I'm doing it on Melodic and it indeed fails there. Need to look why...

rhaschke, 2020-12-08:
Looks like I fixed this back in https://github.com/ros-visualization/rviz/commit/f3eeaf4c695ed6bb1ef990e3cc37fb23801e5c1b for Noetic. However, due to the required API change this will not be backported to Melodic. Please upgrade to rviz' Noetic branch. To this end, you will also need to upgrade [`interactive_markers`](https://github.com/ros-visualization/interactive_markers) to Noetic.

closed 2020-12-08 by rhaschke