ros-visualization/rviz #1569: ROS Info "mesh_use_embedded_materials is ignored."

georgiablanco, 2020-12-16:
When launching our robot arm, we get this ROS info output:
```
[ INFO] [1606930414.061245454, 11.166000000]: Marker 'EE:goal_ra_ee_link/10': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061380945, 11.166000000]: Marker 'EE:goal_ra_ee_link/11': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061457168, 11.166000000]: Marker 'EE:goal_ra_ee_link/12': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061531312, 11.166000000]: Marker 'EE:goal_ra_ee_link/13': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061602528, 11.166000000]: Marker 'EE:goal_ra_ee_link/14': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061685230, 11.166000000]: Marker 'EE:goal_ra_ee_link/15': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061768251, 11.166000000]: Marker 'EE:goal_ra_ee_link/16': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061847196, 11.166000000]: Marker 'EE:goal_ra_ee_link/17': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061926567, 11.166000000]: Marker 'EE:goal_ra_ee_link/18': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.061998879, 11.166000000]: Marker 'EE:goal_ra_ee_link/19': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.062070428, 11.166000000]: Marker 'EE:goal_ra_ee_link/20': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.062150428, 11.166000000]: Marker 'EE:goal_ra_ee_link/21': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.062341263, 11.167000000]: Marker 'EE:goal_ra_ee_link/22': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.062616851, 11.167000000]: Marker 'EE:goal_ra_ee_link/25': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.062696936, 11.167000000]: Marker 'EE:goal_ra_ee_link/26': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.062774298, 11.167000000]: Marker 'EE:goal_ra_ee_link/27': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063018746, 11.167000000]: Marker 'EE:goal_ra_ee_link/32': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063097557, 11.167000000]: Marker 'EE:goal_ra_ee_link/33': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063165193, 11.167000000]: Marker 'EE:goal_ra_ee_link/34': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063347443, 11.167000000]: Marker 'EE:goal_ra_ee_link/38': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063422820, 11.167000000]: Marker 'EE:goal_ra_ee_link/39': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063505026, 11.167000000]: Marker 'EE:goal_ra_ee_link/40': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063722156, 11.167000000]: Marker 'EE:goal_ra_ee_link/44': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063841224, 11.167000000]: Marker 'EE:goal_ra_ee_link/45': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.063945139, 11.168000000]: Marker 'EE:goal_ra_ee_link/46': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.064096101, 11.168000000]: Marker 'EE:goal_ra_ee_link/48': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.064225473, 11.168000000]: Marker 'EE:goal_ra_ee_link/50': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.064332779, 11.168000000]: Marker 'EE:goal_ra_ee_link/51': mesh_use_embedded_materials is ignored.
[ INFO] [1606930414.064432520, 11.168000000]: Marker 'EE:goal_ra_ee_link/52': mesh_use_embedded_materials is ignored.
```

In the cases that our link is just a link as seen [here](https://github.com/shadow-robot/sr_interface/blob/melodic-devel/sr_multi_moveit/sr_multi_moveit_config/config/ur/ur10e.srdf#L16) with no mesh (because we don't need it) is there any way we can not get this info mesage which comes from [here](https://github.com/rhaschke/rviz/blob/melodic-devel/src/rviz/default_plugin/marker_utils.cpp#L395)?

### Environment
* OS Version: Ubuntu 18.04
* ROS Distro: [Melodic]
* RViz, Qt, OGRE, OpenGl version:
```
[ INFO] [1608109652.560032440]: rviz version 1.13.15
[ INFO] [1608109652.560057756]: compiled against Qt version 5.9.5
[ INFO] [1608109652.560064584]: compiled against OGRE version 1.9.0 (Ghadamon)
```

rhaschke, 2020-12-28:
Unfortunately, I cannot reproduce this issue. Could you, please, provide a minimal urdf, srdf, moveit config to reproduce this?
Note, that you don't need to explicitly specify an end-effector (group and end_effector) if you have a simple chain only.

georgiablanco, 2021-01-13:
Hi,

To reproduce the issue you can do so from this repo:
https://github.com/ros-industrial/universal_robot/tree/melodic-devel

when i run 
`roslaunch ur10_moveit_config demo.launch`
I receive ROSINFO of:
`[ INFO] [1610549841.774866635]: Marker 'EE:goal_ee_link/10': mesh_use_embedded_materials is ignored.`

which is the same issue as the above.

rhaschke, 2021-01-14:
Thanks for providing this link. Actually, I was suppressing these warnings from rviz via my `$ROSCONSOLE_CONFIG_FILE`.
Now, I'm able to reproduce them and I filed corresponding fixes in MoveIt.

closed 2021-01-14 by rhaschke

georgiablanco, 2021-01-14:
could you please provide the PR so I can have a look?

rhaschke, 2021-01-14:
Have a look at the latest commits on the melodic-devel or master branches. There is no PR for this simple fix.