ros-visualization/rviz #1571: Accessing PanelDockWidget of custom plugin

Pietroro, 2020-12-17:
Hello everyone, 

I I was wondering whether or not is is possible to access the PanelDockWidget of a custom plugin. The panel seems to be loaded in using the pluginlib_factory, but I am not sure if there is a reference through which I can get to the actual PanelDockWidget object. 

The parent of the rviz::Panel seems to be just an empty reference, so I can't get anywhere with that. Thanks for your help, I appreciate the awesome tool you have put together!

rhaschke, 2020-12-17:
There is getAssociatedWidgetPanel():
https://github.com/ros-visualization/rviz/blob/ebfbb1f5e3281f969966dc9c305b5afc444cb79c/src%2Frviz%2Fdisplay.h#L215-L218

Pietroro, 2020-12-17:
Thanks for the quick reply. 

I may be misunderstanding this, but to call that method I would still need a display object. My plugin is in the form of a panel, not a display, is there something I am missing?

Just to clarify, to reference the panel plugin development tutorial, is there a way to reference the PanelDockWidget (red highlight) from inside the constructor of the panel itself?

[image]

Thanks again for your time

rhaschke, 2020-12-17:
> My plugin is in the form of a panel, not a display

You didn't mention that fact before. :wink:

> Is there a way to reference the PanelDockWidget (red highlight) from inside the constructor of the panel itself?

No, that's not possible. At the time the Panel is created, the corresponding dock widget doesn't even exist.
Why do you need to access the dock widget?

Pietroro, 2020-12-17:
Ah ok, I understand. 

I just wanted to add a "help" icon button next to the close_button, which is in the titleBarWidget, constructed with the PanelDockWidget. I can of course reach the same functionality by adding it in the panel itself. I just thought it would be a lot cleaner to have something that looks roughly like rqt's "panel title bar":

[image]

It's unfortunate that there is no way to gain access to it at the time the panel is created. I have already taken up a lot of your time, but I really guess there is no clean way to do this, is there?

rhaschke, 2020-12-17:
You should be able to access the dock widget (as the parent of the panel) from within `Panel::onInitialize()`. However, adding new tool buttons to the title bar widget, will depend on (private) implementation details of `PanelDockWidget` and thus might break in future.

Pietroro, 2020-12-17:
Thank you again for your help. You were of course right, after the panel has been fully initialized I am able to access its parent. 

I am not sure whether using a QPushButton instead of a QToolButton should make the implementation any sturdier. AFAIK they seem very similar but under the hood they work very differently. Here is my dangerously badly written code for posterity. 

Feel free to shame me for it, I will close the issue then.

```C++
void MyPanel::onInitialize()
{
    // Dynamic cast of parent to PanelDockWidget class, not a good Idea, but we know the parent has to be of that class
    rviz::PanelDockWidget* dock = dynamic_cast<rviz::PanelDockWidget*>(rviz::Panel::parent());
    // Temporarily store the title bar
    QWidget* title = dock->titleBarWidget();
    // Init the button
    QToolButton* help_button = new QToolButton();
    help_button->setIcon(QIcon::fromTheme("help-contents"));
    help_button->setIconSize(QSize(10, 10));
    // Add button to layout, again with the dynamic cast terrible idea
    QHBoxLayout* layout = dynamic_cast<QHBoxLayout*>(title->layout());
    // Insert button second to last
    layout->insertWidget(layout->count()-1,help_button);
}
```

Thanks again for your time

rhaschke, 2020-12-17:
Looks good to me. You are free to use this snippet for your panel plugins, of course. I suggest adding a sanity check before using dock. Maybe the dynamic cast fails in some cases.

Pietroro, 2020-12-17:
Thank you very much, you have been incredibly helpful!

closed 2020-12-17 by Pietroro