ros-visualization/rviz #1575: The plugin for class 'rviz/Orbit' failed to load - ROS Noetic

TheCodeSummoner, 2020-12-25:
### Issue

Can't run `rviz` since *some* recent update (AUR).

The following commands attempt to start `f1tenth` simulator:

```
source ackermann/devel/setup.zsh
source f1tenth/devel/setup.zsh
roslaunch f1tenth_simulator simulator.launch
```

### Output

The following 2 windows open consecutively:

[image]

[image]

The following stream of errors follows in console immediatelly:

```
[ INFO] [1608904938.394390818]: OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
[ERROR] [1608904938.433354113]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Failed to load library /opt/ros/noetic/lib/librviz_default_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/noetic/lib/libtf.so: undefined symbol: _ZN7tf2_ros17TransformListenerC1ERN3tf210BufferCoreERKN3ros10NodeHandleEb)
[ INFO] [1608904938.457024040]: Simulator constructed.
[ERROR] [1608904939.539540543]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: Failed to load library /opt/ros/noetic/lib/librviz_default_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/noetic/lib/libtf.so: undefined symbol: _ZN7tf2_ros17TransformListenerC1ERN3tf210BufferCoreERKN3ros10NodeHandleEb)
[ERROR] [1608904939.541779886]: PluginlibFactory: The plugin for class 'rviz/Interact' failed to load.  Error: Failed to load library /opt/ros/noetic/lib/librviz_default_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/noetic/lib/libtf.so: undefined symbol: _ZN7tf2_ros17TransformListenerC1ERN3tf210BufferCoreERKN3ros10NodeHandleEb)
(...)
```

### Environment

#### OS

- Distributor ID:	ManjaroLinux
- Description:	Manjaro Linux
- Release:	20.2
- Codename:	Nibia

#### ROS

- Noetic
- rviz version 1.14.4
- compiled against Qt version 5.15.2
- compiled against OGRE version 1.9.0 (Ghadamon)

#### Package manager

- yay v10.1.0 - libalpm v12.0.2

rhaschke, 2020-12-28:
This works with standard Ubuntu packages. Check whether you have both `libtf.so` and `libtf2_ros.so` installed.

TheCodeSummoner, 2020-12-28:
This worked a week or two ago, it must be some update related issue. I will try reinstalling ROS later today to see if that's a fix.

I have both libs:
```
~ >>> ls /opt/ros/noetic/lib | grep libtf                                                                                                                                                                         
libtf2_ros.so
libtf2.so
libtf_conversions.so
libtf.so
```

TheCodeSummoner, 2020-12-28:
It seems I can't even install ROS anymore, getting `/usr/bin/ld: /opt/ros/noetic/lib/libtf.so: undefined reference to `tf2_ros::TransformListener::TransformListener(tf2::BufferCore&, ros::NodeHandle const&, bool)'` in multiple packages.

TF2 libraries have been last updated on AUR 2020-12-08 17:18 which would match the sudden "things stopped working" behaviour, therefore the culprit may be somewhere there. Any suggestions?

rhaschke, 2020-12-28:
Installing ROS shouldn't require libtf to work. As you are using a non-standard Linux distro, you should file an issue with your package provider.

TheCodeSummoner, 2020-12-28:
Okay, I will continue investigating what's wrong with the recent update of `libtf.so` on my distro, thank you for your time.

closed 2020-12-28 by TheCodeSummoner

gavanderhoorn, 2021-01-03:
Adding another data point: [rviz doesn't start](https://answers.ros.org/question/368786) on ROS Answers.

Not saying it's the exact same cause, but at least the missing symbol appears to be the same.

marcoarruda, 2021-09-22:
I had exactly the same issue.

I figured out the problem was that I installed **ros 2 foxy** and it configured the variable **$LD_LIBRARY_PATH** to use ROS 2 libraries.

After removing ros 2 workspace, I rebooted my computer and tried an **echo** on this variable

```
home/user/ROS/catkin_ws/devel/lib:/opt/ros/noetic/lib
```

And now rviz works properly