ros-visualization/rviz #1577: Crash on OS X Catalina 10.15.7

zarathustr, 2021-01-18:
* OS Version: OS X Catalina 10.15.7
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz: 
RViz 1.14.0
Qt 5.15.2
OpenGL 2.1 (GLSL 1.2)

The crash report is:
```
Thread 6 Crashed:: Dispatch queue: OpenGLMT
0   libGPUSupportMercury.dylib    	0x00007fff55507983 gpusLoadCurrentVertexArray + 594
1   com.apple.AMDRadeonX6000GLDriver	0x000000012db3bb18 gldUpdateDispatch + 2662
2   GLEngine                      	0x00007fff3ec93fc3 gleDoDrawDispatchCore + 693
3   GLEngine                      	0x00007fff3ec53d0c glDrawArraysInstanced_STD_Exec + 316
4   GLEngine                      	0x00007fff3ec53bc9 glDrawArrays_UnpackThread + 40
5   GLEngine                      	0x00007fff3ec8faa8 gleCmdProcessor + 77
6   libdispatch.dylib             	0x00007fff6e0b6658 _dispatch_client_callout + 8
7   libdispatch.dylib             	0x00007fff6e0bbc44 _dispatch_lane_serial_drain + 597
8   libdispatch.dylib             	0x00007fff6e0bc5d6 _dispatch_lane_invoke + 363
9   libdispatch.dylib             	0x00007fff6e0c5c09 _dispatch_workloop_worker_thread + 596
10  libsystem_pthread.dylib       	0x00007fff6e310a3d _pthread_wqthread + 290
11  libsystem_pthread.dylib       	0x00007fff6e30fb77 start_wqthread + 15
```

It seems that the systems encounters an error during the preparation step of GPU. Can anyone help with this issue?

rhaschke, 2021-02-05:
Unfortunately, there is no active maintainer left running on OSX. Help from the community is highly appreciated.

rhaschke, 2021-02-06:
Duplicate of #1381?

zarathustr, 2021-02-06:
I may have obtained the error source. The reason may have been that the OSX system does not have a supported graphics card. This graphics card should be supported via OpenGL, OpenCL and Metal for OSX basic frameworks. Installing virtual-machined OSX reproduces these errors but they disappeared on every real Mac hardwares available.

rhaschke, 2021-02-06:
Indeed, rviz relies on OpenGL. If this issue can be closed, please do so.

zarathustr, 2021-02-07:
I suggest not to close this issue directly. Because there may be an approach to resolve it. I think this can be fixed by adding some third-party OpenGL libraries, trying now...

Tobias-Fischer, 2021-03-31:
Feel free to check out [RoboStack](https://github.com/RoboStack/ros-noetic) where we have rviz binaries for OS X (and many other ROS packages).

zarathustr, 2022-04-14:
I somehow managed to reproduce the errors on macOS Monterey with Macports installed. The conclusion is that there is no problem with Ogre3D. I fixed this by remove the libGL*.dylib files in /opt/local (prefix of Macports) to elsewhere. I suppose the main reason is that the runtime execution did not find the correct OpenGL library in the DYLD_LIBRARY_PATH.