ros-visualization/rviz #1578: how to use this package?

Joanna-HE, 2021-01-18:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Kinetic | Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    paste your console output here
    ```
* If source build, which git commit?

rhaschke, 2021-01-18:
See the various wikis, e.g. http://wiki.ros.org/rviz/UserGuide or ask a more specific question on https://answers.ros.org.
Please, post real issues and bug reports here only.

closed 2021-01-18 by rhaschke

Joanna-HE, 2021-01-18:
Sorry, I am not familiar with the ROS. I have use "python3 setup.py build" and "python3 setup.py install" to install this package, but it is hard for me to use it for my project.

rhaschke, 2021-01-18:
Please read the docs.