ros-visualization/rviz #1579: inconsistent lighting in Camera plugin

christian-rauch, 2021-01-18:
Describe your issue here and explain how to reproduce it.

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
rviz version 1.14.4
compiled against Qt version 5.12.8
compiled against OGRE version 1.9.0 (Ghadamon)
```

The Camera display type overlays the rendered RViz scene in top of the 2D image. As the rendered scene always represents the current RViz scene, it also includes changes in lighting and the pivot point. Changing the viewpoint in the 3D RViz view will change the lighting in the rendered Camera view.

For consistency, the plugin should render with a constant light source and not change the lighting based on the viewpoint in the 3D view.

rhaschke, 2021-01-19:
I can confirm this issue: The camera display uses the same light source as the main view, where the (unique) light source moves with the camera. However, I'm not sure how to solve this problem, both conceptually and technically: 

1. Which light source should be used for the camera display? Also one coming from the camera?
2. As both renderings share the same scene, is it possible at all to have different light sources in both views?
2. How/where to define this light source?
I think, it would be best if there would be a new panel to manage light sources, configuring their properties like:
    - type: directed vs. ambient
    - color
    - direction (in case of directed light)

christian-rauch, 2021-01-19:
> Which light source should be used for the camera display? Also one coming from the camera?

If the camera display uses a light source, it should be related to the camera frame. It could simply be a surface light behind the camera.

> As both renderings share the same scene, is it possible at all to have different light sources in both views?

I think no. Either the camera view has to be rendered a second time offscreen or you use the same light source for both views.

> How/where to define this light source?

Does the visualisation actually need a light source? Or would it be enough to just have the meshes self-illuminating where the shader basically shows the defined colour, without taking any light source into account?

rhaschke, 2021-01-19:
Without a light source, everything is just black. So, we would need to add an ambient light at least.

christian-rauch, 2021-01-19:
> Without a light source, everything is just black

You can formulate the shader such that it sets the pixel simply to the colour of the mesh. This does not need external lighting, the "light intensity" would be just defined by the colour of the mesh. This would however not work with alpha channels etc.

If these features are required, I think the only way is to render the scene offscreen again with a different light source.

rhaschke, 2021-01-19:
The scene is rendered again anyway (from a different point of view). I wouldn't deviate from directional light sources: they introduce "naturalistic" shading.

christian-rauch, 2021-01-19:
Ok, if it is re-rendered anyway for the Camera Display, then it only needs a constant, camera-frame dependent and viewpoint independent, light source.

rhaschke, 2021-01-19:
It is re-rendered, but currently using the exact same scene, which - as I understand - comprises the light source(s). So, either we need to maintain two similar scenes (only differing in their light sources) or we need to switch light sources before rendering the (same) scene into each view.

simonschmeisser, 2021-02-16:
there is this plugin (and forks) that worked quite well for us when playing around with VR or shadows: https://github.com/mogumbo/rviz_lighting

rhaschke, 2021-03-04:
@christian-rauch, does the referenced plugin solves your problem? Can we close this issue?

christian-rauch, 2021-03-04:
As far as I understand, this plugin allows adding additional light sources to the scene, which is not really what I am interested in. I am just interested in having the light source that already exists to stay fixed with respect to the world reference frame. On top of that, the plugin requires additional patches on top of RViz.

simonschmeisser, 2021-03-04:
yes, you can disable the default one and add a new fixed on using the plugin I linked to. (or maybe one of the forks)

The patches discussed there would give more realistic (a little bit at least) lighting but just to have a fixed light source you don't need any fixes AFAIR

closed 2021-06-18 by rhaschke

christian-rauch, 2021-06-18:
Was this fixed upstream?

rhaschke, 2021-06-18:
This has gotten the label `wontfix`. Simon suggested a plugin instead.

christian-rauch, 2021-06-18:
The plugin needs a custom patch applied to RViz (https://github.com/mogumbo/rviz_lighting#making-it-work-better-with-rviz). Can we apply this patch at least, so that the plugin can be used straightforward without requiring to compile RViz manually?

rhaschke, 2021-06-18:
> I am just interested in having the light source that already exists to stay fixed with respect to the world reference frame.

Fixating the light source isn't a great idea, because you won't see anything than black if looking against the light.

> Can we apply this patch at least, so that the plugin can be used straightforward without requiring to compile RViz manually?

Disabling the default light source (as done by the patch) isn't a good idea either, because you won't see anything by default.
@christian-rauch: If you come up with a good solution to these problems, I'm happy to consider your PRs.

simonschmeisser, 2021-06-18:
Isn't there even a checkbox for disabling global ligthing already?
The parts that chance attenuation and co I haven't tested yet but they are optional improvements (I'll try to understand and test those soonish)