ros-visualization/rviz #1580: rviz segmentation fault on arch linux melodic

jwhendy, 2021-01-19:
I'm on arch linux and recently we had some changes come through (boost and python) which required rebuilding my ROS melodic install. I'm now getting seg faults with rviz. I'm not sure how to track it down further to understand what the cause is.
```
$ rviz
[ INFO] [1611020873.593200465]: rviz version 1.13.15
[ INFO] [1611020873.593226767]: compiled against Qt version 5.15.2
[ INFO] [1611020873.593239262]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1611020878.529754532]: Forcing OpenGl version 0.
[ INFO] [1611020878.998484174]: Stereo is NOT SUPPORTED
[ INFO] [1611020878.998529931]: OpenGl version: 4.6 (GLSL 4.6).
Segmentation fault (core dumped)
```
[Here's](https://gist.github.com/jwhendy/b8ea5babf8a7cd8cd793075eb30b1f05) the `gdb` output, though I admit I have little familiarity with it. I compiled with `-DCMAKE_BUILD_TYPE=Debug` but I still see "no debugging symbols found" in the output? If I should be doing something else to provide more helpful output, could you kindly let me know option to use?

### Your environment
* OS Version: 
```
$ uname -a
Linux 5.10.7-arch1-1 #1 SMP PREEMPT Wed, 13 Jan 2021 12:02:01 +0000 x86_64 GNU/Linux
```
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz: see above
* building from this [arch repo](https://github.com/ros-melodic-arch/ros-melodic-rviz)

Thanks for taking a look.

rhaschke, 2021-01-19:
I never have seen issues in `Ogre::ScriptCompiler::setListener` before. Hence, I'm assuming a broken build environment: as you have changes to python and boost, maybe parts of the software still uses outdated boost headers?

jwhendy, 2021-01-19:
Thanks for the gut check. I went on a bit of a hunt, guessing at boost being a culprit as we recently implemented some "tricks" to use `boost-1.69` instead of the system installed `boost` which regularly updates (currently at 1.75). I tracked down a double dependency via `ldd` of `rviz` on both 1.69 and 1.75 variants and this tracked to `ogre`. Linked to our arch issue tracker above for reference! Thanks again.

closed 2021-01-19 by jwhendy

AlexanderDavid, 2021-02-04:
@jwhendy Have you found a fix for this? I found that I am having the same issue.

jwhendy, 2021-02-04:
@AlexanderDavid perhaps at least a partial lead on one. Did you take a look at the linked issue above from the ros-melodic-arch? Here's the [relevant comment](https://github.com/ros-melodic-arch/ros-melodic-rviz/issues/12#issuecomment-763072693) where I think my issue was coming from conflicting boost versions.

Feel free to join in that conversation if it's relevant. You'd be looking for `ldd` output of `/opt/ros/melodic/bin/rviz` to see if it's linked to two different versions. As for the complete fix, we're still working that out. We want to use `boost1.69` in `/opt` for ROS so that we don't have to wrestle with Arch updating much more frequently.

jwhendy, 2021-02-04:
@AlexanderDavid oh, and if you don't catch it in that thread, ogre was also a culprit as rviz relies on it (though the root cause was still sorting out boost library deps).

AlexanderDavid, 2021-02-04:
@jwhendy I saw the thread but didn't see a fix. I just added the oscloud repo to pacman and installed everything from there to fix my issue. But I guess I didn't fix the root cause.

jwhendy, 2021-02-04:
@AlexanderDavid the [partial] fix for me was rebuilding ogre (which didn't have our boost1.69 cmake arg as it's not in our repo) and then rebuilding rviz. The full fix seems to require building in a clean chroot to avoid linking to both boost version. I'm not sure if Oscar has done that with his repo.