ros-visualization/rviz #1581: RViz shows unintentional reflected planes from point cloud

Martzi, 2021-01-21:
### My environment
* OS Version:Ubuntu 18.04 (kernel 5.4.0-62-generic)
* ROS Distro: Melodic
* RViz version 1.13.15
* Qt version 5.9.5 
* OGRE version 1.9.0 (Ghadamon)
* OpenGl version: 4,6 (GLSL 4,6)

### My goal
Streaming point cloud captured by Intel RealSense d435i camera from a linux pc through tcp, then convert this data and publish in a ROS pointcloud2 topic, then visualize in RViz.

### My issue
I try to visualize pointcloud2 data from an Intel RealSense d435i camera. First, I tried with random generated numpy data with open3D (v0.10.0.0) to check wether my script is able to show up in RViz. It seemd to be woring. I checked with open3D's viewer, which was also worked.
Here comes the interesting part....
I tried these scripts with the d435i camera. If I view the received pointcloud with open3D viewer, seems fine. I created a ROS node to convert this numpy array to a pointcloud2 data, then RViz reflects data to 2 different coordinate planes. I attach images below. There you can see the point cloud visualized by open3D and from RViz. RViz shows duplicates of an original point cloud, but why? could you help me out?
[this is my function](https://gist.github.com/Martzi/90d8943a93cb91fd54327f5f1602450b) to create a pointcloud2 message
Here are the images from RViz with the plane reflections, and the correct one from open3D viewer

[image]
[image]
[image]

rhaschke, 2021-01-21:
Instead of testing your script with random data (which is hard to validate), use some simple geometric shapes, e.g. a circle, box, or sphere of points. rviz, for sure, is not to blame for this issue.

closed 2021-01-21 by rhaschke

Martzi, 2021-01-22:
I tested with a sphere generated with open3D and viewed with RViz. It displayed the sphere correctly.
Now the problem is reducated to the pointcloud coming from the d435i camera (and still RViz...). How is it possible that open3D shows it correctly meanwhile RViz not? This meant to demonstrate the attached images. I think if my pontcloud generation were wrong, open3D should also showed it wrong, didn't it?

rhaschke, 2021-01-22:
Your script uses different dtype and itemsize definitions. Are you sure, the point cloud returned by the camera is float32 for all components?
Try to simplify the point cloud to 2-3 points only and verify the generated ROS msg only includes those points (i.e. not mirroring them onto themselves).

Martzi, 2021-01-22:
I uploaded my full project here: [https://github.com/Martzi/pcl-to-ros](https://github.com/Martzi/pcl-to-ros)
If you start the pc_pub_gen.py, it sends the generated data through localhost to be able to test locally with my node. Now I see that the failure is somewhere in my pointcloud2 publisher (publish_pc2), but I still could not manage to correct it :(. I recalculated the PointField values, but still wrong somewhere.

rhaschke, 2021-01-22:
Sorry, I cannot help any further. This question should be better asked at https://answers.ros.org.
Please, post real issues and bug reports here only.

Martzi, 2021-01-22:
all right, thanks for your help!