ros-visualization/rviz #1582: segfault on ubuntu MATE

wydmynd, 2021-01-26:
### Your environment
* OS Version: e.g. Ubuntu MATE 20.04.01
* ROS Distro: [Noetic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:

```
    [ INFO] [1611643030.717324667]: rviz version 1.14.4
[ INFO] [1611643030.717521614]: compiled against Qt version 5.12.8
[ INFO] [1611643030.717591226]: compiled against OGRE version 1.9.0 (Ghadamon)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
No XVisualInfo for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Falling back to using screens root_visual.
[ INFO] [1611643039.081698617]: Forcing OpenGl version 0.
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
No XVisualInfo for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Falling back to using screens root_visual.
Segmentation fault (core dumped)

```
solved by - 
sudo apt-get install libgl1-mesa-dri

rhaschke, 2021-01-26:
Obviously, one needs to have a working OpenGL config. This is usually handled by the dependency hierarchy, which seems to be broken on MATE?

closed 2021-02-05 by rhaschke

wydmynd, 2021-02-06:
just trying to clarify, I'm not a linux wizard, just a beginner trying to install rviz. 
it took me hours of googling to find the solution (after trying many other things)
my point is that maybe the software should not just crash? but notify me of the problem?

rhaschke, 2021-02-06:
Sure. But, if we cannot reproduce the issue locally, it's hard to fix it.