ros-visualization/rviz #1584: rviz segfaults on pointclou2 with new opengl

SquirrelCZE, 2021-02-05:
Got segfault with pointcloud2 view from realsense camera.
Everytime I try to enable the plugin for it and receive data (works in case it has no topic), rviz immediatly segfaults.

Everything works just OK with '--opengl 210'.

### Your environment
* OS Version: e.g. Ubuntu 20.4
* ROS Distro: Melodic
* GPU/Processor: AMD Ryzen 7 PRO 3700U w/ Radeon Vega Mobile Gfx
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
  [ INFO] [1612557068.837319690]: rviz version 1.14.4
  [ INFO] [1612557068.837440765]: compiled against Qt version 5.12.8
  [ INFO] [1612557068.837469989]: compiled against OGRE version 1.9.0 (Ghadamon)
  [ INFO] [1612557068.854635875]: Forcing OpenGl version 0.
  [ INFO] [1612557069.359404664]: Stereo is NOT SUPPORTED
  [ INFO] [1612557069.359574930]: OpenGl version: 4,6 (GLSL 4,6).
* Ogre log:
  [ INFO] [1612557227.031757660]: GLRenderSystem::_createRenderWindow "OgreWindow(1)", 100x30 windowed  miscParams: FSAA=4 contentScalingFactor=1,000000 externalGLControl= externalWindowHandle=41943063 parentWindowHandle=41943063 
[ INFO] [1612557227.075792334]: GLXWindow::create used FBConfigID = 605
[ INFO] [1612557227.217623929]: Mesh: Loading rviz_sphere.mesh.
[ INFO] [1612557227.223506398]: Texture: SelectionRect0Texture: Loading 1 faces(PF_R8G8B8A8,1x1x1) with 0 generated mipmaps from Image. Internal format is PF_A8R8G8B8,1x1x1.
[ INFO] [1612557227.225629624]: Mesh: Loading rviz_cylinder.mesh.
[ INFO] [1612557227.225906850]: Mesh: Loading rviz_cone.mesh.
[ INFO] [1612557227.314088158]: Using FSAA from GL_ARB_multisample extension.
[1]    138464 segmentation fault (core dumped)  /opt/ros/noetic/lib/rviz/rviz -l
* Backtrace
(gdb) backtrace
#0  0x0000000000000000 in ?? ()
#1  0x00007fff4ef04a3d in Ogre::GLSL::GLSLLinkProgram::compileAndLink() ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#2  0x00007fff4ef05796 in Ogre::GLSL::GLSLLinkProgram::activate() ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#3  0x00007fff4ef07de5 in Ogre::GLSL::GLSLLinkProgramManager::getActiveLinkProgram() ()
   from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#4  0x00007fff4ef030c1 in Ogre::GLSL::GLSLGpuProgram::bindProgramParameters(Ogre::SharedPtr<Ogre::GpuProgramParameters>, unsigned short) () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#5  0x00007fff4ef58291 in Ogre::GLRenderSystem::bindGpuProgramParameters(Ogre::GpuProgramType, Ogre::SharedPtr<Ogre::GpuProgramParameters>, unsigned short) () from /usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so.1.9.0
#6  0x00007ffff654c694 in Ogre::SceneManager::updateGpuProgramParameters(Ogre::Pass const*) ()
   from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#7  0x00007ffff655412d in Ogre::SceneManager::renderSingleObject(Ogre::Renderable*, Ogre::Pass const*, bool, bool, Ogre::HashedVector<Ogre::Light*> const*) () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#8  0x00007ffff6504679 in Ogre::QueuedRenderableCollection::acceptVisitorGrouped(Ogre::QueuedRenderableVisitor*) const ()
   from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#9  0x00007ffff650478c in Ogre::QueuedRenderableCollection::acceptVisitor(Ogre::QueuedRenderableVisitor*, Ogre::QueuedRenderableCollection::OrganisationMode) const () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#10 0x00007ffff654693a in Ogre::SceneManager::renderBasicQueueGroupObjects(Ogre::RenderQueueGroup*, Ogre::QueuedRenderableCollection::OrganisationMode) () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#11 0x00007ffff65467d7 in Ogre::SceneManager::renderVisibleObjectsDefaultSequence() () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#12 0x00007ffff65584a7 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) ()
   from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#13 0x00007ffff6392c3a in Ogre::Camera::_renderScene(Ogre::Viewport*, bool) () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#14 0x00007ffff651e805 in Ogre::RenderTarget::_updateViewport(Ogre::Viewport*, bool) ()
   from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#15 0x00007ffff651e612 in Ogre::RenderTarget::_updateAutoUpdatedViewports(bool) () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#16 0x00007ffff651e363 in Ogre::RenderTarget::updateImpl() () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#17 0x00007ffff651ed18 in Ogre::RenderTarget::update(bool) () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#18 0x00007ffff650890f in Ogre::RenderSystem::_updateAllRenderTargets(bool) () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#19 0x00007ffff653f51e in Ogre::Root::_updateAllRenderTargets() () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#20 0x00007ffff653f610 in Ogre::Root::renderOneFrame() () from /lib/x86_64-linux-gnu/libOgreMain.so.1.9.0
#21 0x00007ffff7f3f570 in rviz::VisualizationManager::onUpdate() () from /opt/ros/noetic/lib/librviz.so
#22 0x00007ffff73cc300 in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007ffff73d951e in QTimer::timeout(QTimer::QPrivateSignal) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff73cccf5 in QObject::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007ffff77cda66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff77d70f0 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff73a093a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff73f78b0 in QTimerInfoList::activateTimers() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff73f819c in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00007ffff461ef9d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff461f220 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff461f2c3 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#33 0x00007ffff73f8565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#34 0x00007ffff739f4db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#35 0x00007ffff73a7246 in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x000055555555860d in main ()
* Core file -> it has 571M, not sure how to attach it....

rhaschke, 2021-02-06:
This seems similar to #1508, but in your case an AMD GPU is affected. Can you provide the output of `glxinfo | grep OpenGL` please?

SquirrelCZE, 2021-02-06:
sure:

➜ glxinfo | grep OpenGL             
OpenGL vendor string: AMD
OpenGL renderer string: AMD RAVEN (DRM 3.38.0, 5.8.0-41-generic, LLVM 11.0.1)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.3.4 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 20.3.4 - kisak-mesa PPA
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.3.4 - kisak-mesa PPA
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

rhaschke, 2021-02-08:
I have pushed a new commit to melodic-devel to have rviz report the employ rendering device during startup.
Could you please build the latest melodic-devel branch please from source and report this new information?
I'm surprised, that rviz doesn't automatically downgrade to an older OpenGL version, because we know that the Mesa libs have issues rendering rviz' more modern material scripts.

SquirrelCZE, 2021-02-09:
I tried to just compile it localy in workspace, btu got an error:

/home/veverak/Projects/Schpin/repos/thirdparty/rviz/src/rviz/default_plugin/interactive_marker_display.cpp:122:100: error: no matching function for call to ‘interactive_markers::InteractiveMarkerClient::InteractiveMarkerClient(tf::Transformer&, std::string)’

using everything as packages from repository, any tips how to make this?

rhaschke, 2021-02-09:
This looks like a mismatch between Melodic and Noetic. Which system are you compiling on? Which branch did you checkout?

SquirrelCZE, 2021-02-10:
My mistake, I am no noetic, completely forgot about that

SquirrelCZE, 2021-02-11:
SO, I applied the patch on noetic version which I am using (again, sorrt for that):

[ INFO] [1613002034.041585762]: rviz version 1.14.4
[ INFO] [1613002034.041681030]: compiled against Qt version 5.12.8
[ INFO] [1613002034.041716486]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1613002034.062530782]: Forcing OpenGl version 0.
[ INFO] [1613002034.436060548]: Stereo is NOT SUPPORTED
[ INFO] [1613002034.436311486]: OpenGL device: AMD RAVEN (DRM 3.38.0, 5.8.0-43-generic, LLVM 11.0.1)
[ INFO] [1613002034.436391325]: OpenGl version: 4,6 (GLSL 4,6).

rhaschke, 2021-02-11:
Thanks for the feedback. Could you please insert the following line of code:
```c
std::cerr << glGetString(GL_VENDOR) << " - " << glGetString(GL_RENDERER) << " - " << glGetString(GL_VERSION) << std::endl;
```

into the function `RenderSystem::detectGlVersion` such that this info is printed on startup as well?
https://github.com/ros-visualization/rviz/blob/c36e265d9954a3a7d4a40f1db0821e5ce1a10c15/src/rviz/ogre_helpers/render_system.cpp#L170

SquirrelCZE, 2021-02-11:
AMD - AMD RAVEN (DRM 3.38.0, 5.8.0-43-generic, LLVM 11.0.1) - 4.6 (Compatibility Profile) Mesa 20.3.4 - kisak-mesa PPA

closed 2021-02-11 by rhaschke

rhaschke, 2021-02-17:
> Everything works just OK with '--opengl 210'.

@SquirrelCZE, could you please check whether '--opengl 310' resolves the segfault as well?

rhaschke, 2021-03-26:
Dear @SquirrelCZE,
could you please confirm whether you face https://github.com/ros-visualization/rviz/issues/1508#issuecomment-801084689l as well on your AMD hardware? I filed an issue on Mesa for that and they are wondering which type of hardware is affected.

PR #1588 (2021-02-11): Enforce GLSL 1.4 on more Mesa systems (#1559)
Mesa-based rendering issues showed up e.g. on AMD GPU as well. The new approach should handle any Mesa 20-based system.
Fixes #1584.