ros-visualization/rviz #1585: Deleting all markers in a MarkerArray topic

seann999, 2021-02-09:
I am trying to delete all markers in a MarkerArray topic but rviz is not reflecting this. I could not find documentation on specifically how to do this, but below is how I am trying to do it:
```
markers = MarkerArray()
marker = Marker()
marker.header.frame_id = 'map'
marker.action = marker.DELETEALL
markers.markers.append(marker)
self.attempt_pub.publish(markers)
```
However, the markers in attempt_pub are not getting removed. Am I doing something wrong?

### Your environment
* OS Version: e.g. Ubuntu 16.04
* ROS Distro: Kinetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1612873226.540147533]: rviz version 1.12.17
    [ INFO] [1612873226.540198650]: compiled against Qt version 5.5.1
    [ INFO] [1612873226.540211896]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1612873227.421703813]: Stereo is NOT SUPPORTED
    [ INFO] [1612873227.421863934]: OpenGl version: 4.6 (GLSL 4.6).
    ```

seann999, 2021-02-09:
Okay, doing the above in the update loop instead of the initialization function made it work.
(solved?)

rhaschke, 2021-02-09:
I'm not sure where the above Python(?) code is embedded. Probably not in rviz. My guess from your sparse info is that your Python program didn't really have time to send out the message before closing. You need to call rospy.Rate(10).sleep() to give Python some time to actually perform all the ROS communication in the background.

closed 2021-03-04 by rhaschke