ros-visualization/rviz #1589: Unhandled exception when marker contains mesh from unknown package

simonschmeisser, 2021-02-16:
When a marker msgs with a mesh from an unknown package is loaded, it throws an Ogre exception which is unfortunately not handled:
```
[ WARN] [1613474597.318764360]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://motek_demo_2017/meshes/tilted-base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/o
gre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)                                                                                                                                                            
terminate called after throwing an instance of 'Ogre::FileNotFoundException'

  what():  OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://motek_demo_2017/meshes/tilted-base.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/ogre-1.9-B6QkmW/ogre-1
.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 756)

backtrace:

1   __GI_raise                                                                                                                                                                                                                                                                     raise.c 51 0x7f6ae9cadfb7 
2   __GI_abort                                                                                                                                                                                                                                                                     abort.c 79 0x7f6ae9caf921 
3   ??                                                                                                                                                                                                                                                                                        0x7f6aea6a2957 
4   ??                                                                                                                                                                                                                                                                                        0x7f6aea6a8ae6 
5   std::terminate()                                                                                                                                                                                                                                                                          0x7f6aea6a8b21 
6   __cxa_rethrow                                                                                                                                                                                                                                                                             0x7f6aea6a8da9 
7   Ogre::Resource::load(bool)                                                                                                                                                                                                                                                                0x7f6ae717fb08 
8   Ogre::MeshManager::load(std::string const&, std::string const&, Ogre::HardwareBuffer::Usage, Ogre::HardwareBuffer::Usage, bool, bool)                                                                                                                                                     0x7f6ae70ec0a7 
9   Ogre::EntityFactory::createInstanceImpl(std::string const&, std::map<std::string, std::string, std::less<std::string>, Ogre::STLAllocator<std::pair<std::string const, std::string>, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0>>> const *)                                     0x7f6ae7013ce6 
10  Ogre::MovableObjectFactory::createInstance(std::string const&, Ogre::SceneManager *, std::map<std::string, std::string, std::less<std::string>, Ogre::STLAllocator<std::pair<std::string const, std::string>, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0>>> const *)            0x7f6ae70fbd44 
11  Ogre::SceneManager::createMovableObject(std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, Ogre::STLAllocator<std::pair<std::string const, std::string>, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0>>> const *)                 0x7f6ae71c477f 
12  Ogre::SceneManager::createEntity(std::string const&, std::string const&, std::string const&)                                                                                                                                                                                              0x7f6ae71c0f1a 
13  rviz::MeshResourceMarker::onNewMessage(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void>> const> const&, boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void>> const> const&)                                                                              0x7f6adba6df9a 
14  rviz::MarkerBase::setMessage(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void>> const> const&)                                                                                                                                                                           0x7f6adba6b571 
15  rviz::MarkerDisplay::processAdd(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void>> const> const&)                                                                                                                                                                        0x7f6adba559e6 
16  rviz::MarkerDisplay::processMessage(boost::shared_ptr<visualization_msgs::Marker_<std::allocator<void>> const> const&)                                                                                                                                                                    0x7f6adba56103 
17  rviz::MarkerDisplay::update(float, float)                                                                                                                                                                                                                                                 0x7f6adba5651f 
18  rviz::DisplayGroup::update(float, float)                                                                                                                                                                                                                                                  0x7f6aed181b88 
19  rviz::VisualizationManager::onUpdate()                                                                                                                                                                                                                                                    0x7f6aed24bb25 
20  QMetaObject::activate(QObject *, int, int, void * *)                                                                                                                                                                                                                                      0x7f6af0631305 
... <Mehr>                                                                                                                                                                                                                                                                                                   

```

### Your environment
* OS Version: e.g. Ubuntu 18.04
* ROS Distro: [Kinetic | Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
    [ INFO] [1613475494.475253896]: rviz version 1.13.15
[ INFO] [1613475494.475289937]: compiled against Qt version 5.12.8
[ INFO] [1613475494.475299717]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1613475494.483962728]: Forcing OpenGl version 0.
[ INFO] [1613475495.046200133]: Stereo is NOT SUPPORTED
[ INFO] [1613475495.046340502]: OpenGl version: 3,0 (GLSL 1,3).

    ```
* If source build, which git commit? d4603a4f

I'll try to provide a PR soon

rhaschke, 2021-02-16:
> I'll try to provide a PR soon

Thanks.

rhaschke, 2021-03-04:
I cannot reproduce the issue. In both, Melodic and Noetic, the unknown file/package is correctly reported in the display status:

[image]

simonschmeisser, 2021-03-15:
Strange one ... I also couldn't reproduce it in standalone rviz but it successfully crashes our UI that includes the `rviz::RenderPanel`

The crash also happens in `InteractiveMarker` 

comparing rviz and the embedded use case I see that rviz prints out the following for every marker, interactive marker, robot:
```
[rospack] Error: package 'motek_demo_2017' not found
[librospack]: error while executing command
[ERROR] [1615826641.080096501]: Could not load resource [package://motek_demo_2017/meshes/tilted-base.stl]: Unable to open file "package://motek_demo_2017/meshes/tilted-base.stl".

```

while rviz embedded in our application only prints that for `RobotStateDisplay` but not for the interactive marker where it crashes. So I wonder why this does not fail:
https://github.com/ros-visualization/rviz/blob/ba536b2d5baae4be167f9741a3f5bcb05d129af6/src/rviz/default_plugin/markers/mesh_resource_marker.cpp#L111

Here is another backtrace:
```
1  __GI_raise                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 raise.c      51  0x7f817c605fb7 
2  __GI_abort                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 abort.c      79  0x7f817c607921 
3  ??                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          0x7f817cffa957 
4  ??                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          0x7f817d000ae6 
5  std::terminate()                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            0x7f817d000b21 
6  __cxa_rethrow                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               0x7f817d000da9 
7  Ogre::Resource::load(bool)                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  0x7f8179ad7b08 
8  Ogre::MeshManager::load(std::string const&, std::string const&, Ogre::HardwareBuffer::Usage, Ogre::HardwareBuffer::Usage, bool, bool)                                                                                                                                                                     [... cut ...]                                                                                                                                                                          0x561bb342435a 
```

I'm not sure how to debug this further and have some headache due to covid so I'd be grateful for ideas. I noticed that `InteractiveMarker` just silently ignores (from a GUI point of view) the broken markers but still it does not crash in rviz. I could not find a general purpose `catch` that would explain it.

rhaschke, 2021-03-15:
I might have an idea: Do you load the same mesh, both with `rviz::Robot` and `InteractiveMarker`? 
The `Robot` code currently ignores the return value of `loadMeshFromResource()` (which should be fixed):
https://github.com/ros-visualization/rviz/blob/13d061d973e96c6315c7768a7f316d2cc501299d/src/rviz/robot/robot_link.cpp#L596-L600 
and thus creates an invalid entity. Next time when the resource is requested, this _existing_ but invalid entity is returned:
https://github.com/ros-visualization/rviz/blob/13d061d973e96c6315c7768a7f316d2cc501299d/src/rviz/mesh_loader.cpp#L671-L674

simonschmeisser, 2021-03-16:
Good catch @rhaschke ! Now I can crash rviz as well (by enabling a `PlanningSceneDisplay` containing the same mesh)

Not setting the entity in `RobotLink` fixes the crash. Will open a PR later, maybe also with some more error reporting in the UI

simonschmeisser, 2021-03-16:
Do you see a more convenient way than adding a chain of signal/slots in `RobotLink`, `Robot` and `RobotModelDisplay` for actually making those resource loading errors visible in an UI/the status field? Same question for `InteractiveMarkers` where it looks even more complicated unfortunately

rhaschke, 2021-03-17:
No, I'm afraid that's the only way. I'm fine to merge #1596 as is.

rhaschke, 2021-06-18:
Closed via #1629.

PR #1629 (2021-06-11): Robot: Report mesh loading issues
This PR implements https://github.com/ros-visualization/rviz/pull/1597#issuecomment-802670884 and fixes #1589.
@simonschmeisser, please have a look. I decided to summarize all mesh loading issues as part of the URDF field.
As you noticed in #1597, TF updates to the link status will overwrite previous errors and introducing a new status field, I didn't like so much.
[image]
closed 2021-06-18 by rhaschke