ros-visualization/rviz #1591: Segfault on startup

kristenmichaelson, 2021-03-02:
Can't run rviz after recent upgrade to Ubuntu 20.04 and ROS Noetic on a virtual machine. Tried changing to X11 window manager, installing mesa libraries, and installing libyaml-cpp-dev. No luck. 

Output:
[ INFO] [1614705317.365471181]: rviz version 1.14.4
[ INFO] [1614705317.365537413]: compiled against Qt version 5.12.8
[ INFO] [1614705317.365547738]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1614705317.375156733]: Forcing OpenGl version 0.
[1]    2462 segmentation fault (core dumped)  rviz

Ran with gdb, and got the following error: 
Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00007ffff7f50098 in rviz::YamlConfigReader::readYamlNode(rviz::Config&, YAML::Node const&) () from /opt/ros/noetic/lib/librviz.so

rhaschke, 2021-03-03:
Do you have any non-standard libs installed in your system? I cannot reproduce this error.
Could you build rviz from source and with debug symbols, please, and report the source-code line failing?

kristenmichaelson, 2021-03-08:
Thank you so much! I don't believe I have any nonstandard libs. It might be worth noting that I recently built another package on this VM (using cmake, not catkin) that has an older version of yaml-cpp as a dependency. Could that cause a problem? 

I cloned the rviz repository in a new catkin_ws/src folder and added set(CMAKE_BUILD_TYPE Debug) to CMakeLists.txt. Then I ran catkin_make, and I got this error:

/home/aerial-robotics/catkin_ws/src/rviz/src/rviz/default_plugin/interactive_marker_display.cpp: In member function ‘virtual void rviz::InteractiveMarkerDisplay::onInitialize()’:
/home/aerial-robotics/catkin_ws/src/rviz/src/rviz/default_plugin/interactive_marker_display.cpp:122:100: error: no matching function for call to ‘interactive_markers::InteractiveMarkerClient::InteractiveMarkerClient(tf::Transformer&, std::string)’
  122 | kers::InteractiveMarkerClient(*tf, fixed_frame_.toStdString()));
      |                                                              ^

In file included from /home/aerial-robotics/catkin_ws/src/rviz/src/rviz/default_plugin/interactive_marker_display.h:43,
                 from /home/aerial-robotics/catkin_ws/src/rviz/src/rviz/default_plugin/interactive_marker_display.cpp:40:
/opt/ros/noetic/include/interactive_markers/interactive_marker_client.h:90:3: note: candidate: ‘interactive_markers::InteractiveMarkerClient::InteractiveMarkerClient(tf2_ros::Buffer&, const string&, const string&)’
   90 |   InteractiveMarkerClient(tf2_ros::Buffer &tf,
      |   ^~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/noetic/include/interactive_markers/interactive_marker_client.h:90:44: note:   no known conversion for argument 1 from ‘tf::Transformer’ to ‘tf2_ros::Buffer&’
   90 |   InteractiveMarkerClient(tf2_ros::Buffer &tf,
      |                           ~~~~~~~~~~~~~~~~~^~
make[2]: *** [rviz/src/rviz/default_plugin/CMakeFiles/rviz_default_plugin.dir/build.make:245: rviz/src/rviz/default_plugin/CMakeFiles/rviz_default_plugin.dir/interactive_marker_display.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:3197: rviz/src/rviz/default_plugin/CMakeFiles/rviz_default_plugin.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed

rhaschke, 2021-03-09:
> I recently built another package on this VM (using cmake, not catkin) that has an older version of yaml-cpp as a dependency. Could that cause a problem?

Sure, that might be the root cause of your problem. Are you using the binary rviz release? If so, check, which libyaml-cpp is used:
```
$ ldd /opt/ros/noetic/lib/librviz.so |grep yaml
	libyaml-cpp.so.0.6 => /lib/x86_64-linux-gnu/libyaml-cpp.so.0.6 (0x00007f4dc6d7c000)
```

Regarding your build problem: Checkout the noetic-devel branch, not melodic-devel ;-)

kristenmichaelson, 2021-03-11:
Right on both counts! Installation worked after checking out the correct branch. Had a bad path in .bashrc from an old fix no longer required after upgrade. Thanks so much for all your help.

closed 2021-03-11 by kristenmichaelson