ros-visualization/rviz #1592: Rviz wrench display parameters missing

Yeshasvitvs, 2021-03-03:
We use Rviz a part of [human-dynamics-estimation](https://github.com/robotology/human-dynamics-estimation) in order to visualize the human motion along with the visualization of ground reaction forces (through WrenchStamped ros messages).  Please take a look at https://www.youtube.com/watch?v=VG5lgNbkpus&ab_channel=IstitutoItalianodiTecnologia for further details.

We have been using ros melodic for a while on Ubuntu 18.04 LTS, and recently updated to use ros noetic along with Ubuntu 20.04  LTS. After some recent updates we started observing a strange behavior in the wrench display as following

https://user-images.githubusercontent.com/6505998/109789740-ed732780-7c10-11eb-9c5e-87dd47cf93f5.mp4

and noticed that some of the wrench display parameters such as force scaling factor and torque scaling factor are missing.

[image]

The usual wrench display parameters before were 

[image]

In order to identify the exact problem, we investigated rviz installation from sources following  http://wiki.ros.org/Installation/Source. Here I replaced rviz ros package from the [rviz upstream noetid-devel branch](https://github.com/ros-visualization/rviz/tree/noetic-devel). To compile the ros workspace, I also had to use [interactive_markers upstream kinetic-devel branch](https://github.com/ros-visualization/interactive_markers/tree/kinetic-devel)  (as there were some errors in compiling with noetic-devel branch).

Under the above setup, I was able to pinpoint the problem to the commit that refactors wrench display to screw display https://github.com/ros-visualization/rviz/commit/dd1779e78dbb1468fdb11968632fafd5e7a272f2. Using the commit before the refactoring of wrench display to screw display, we are able to visualize wrench correctly as before with all the parameters for rviz wrench display

https://user-images.githubusercontent.com/6505998/109792126-79864e80-7c13-11eb-84d1-55e7514aa99e.mp4

We are using the an [rviz configuration file](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/ros/rviz/HDERviz.rviz) that was created from a previous version of rviz, and I see that the wrench display class is [rviz/WrenchStamped](https://github.com/robotology/human-dynamics-estimation/blob/master/conf/ros/rviz/HDERviz.rviz#L723). From the commit in https://github.com/ros-visualization/rviz/commit/dd1779e78dbb1468fdb11968632fafd5e7a272f2 I see this wrench display class has not been moved, but may be the old configuration file is not longer compatible to be used ?

@mvieth @rhaschke please take a look at this issue and let me know if we need to update any configuration parameters. Thanks
 

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic]
* RViz: https://github.com/ros-visualization/rviz/tree/noetic-devel
*  Qt, OGRE, OpenGl version as printed by rviz:

    ```
    [ INFO] [1614767611.288377184]: rviz version 1.13.13
    [ INFO] [1614767611.288459592]: compiled against Qt version 5.12.8
    [ INFO] [1614767611.288486932]: compiled against OGRE version 1.9.0 (Ghadamon)
    [ INFO] [1614767611.295486545]: Forcing OpenGl version 0.
    [ INFO] [1614767611.877200955]: Stereo is NOT SUPPORTED
    [ INFO] [1614767611.877264211]: OpenGl version: 4.6 (GLSL 4.6).

    ```
* If source build, which git commit? https://github.com/ros-visualization/rviz/commit/dd1779e78dbb1468fdb11968632fafd5e7a272f2

mvieth, 2021-03-03:
Hm, I tested the original changes from my pull request (#1506) which seemed to work fine, but the actually committed changes are different from my proposal. I now tested the current master:
* the problem that the display parameters/properties do not appear in the RViz window can be solved by using the full `Property` constructor instead of `QObject::connect` in the `ScrewDisplay` constructor
* This message appears when adding the display (not sure if warning or error): `QObject::connect: No such slot rviz::_RosTopicDisplay::updateProperties()`. I think this can be solved by marking `updateProperties` with `Q_SLOTS`, but that is not possible in `ScrewDisplay` (since templated), only in `[Accel/Twist/Wrench]StampedDisplay`
* RViz crashes with a symbol lookup error: `*processMessagePrivate*` is undefined. This can be solved by instantiating `ScrewDisplay` like this: `template class rviz::ScrewDisplay<geometry_msgs::AccelStamped>;`

There may be other possible solutions, these are just the ones I originally used in my pull request.

rhaschke, 2021-03-04:
Thanks, @Yeshasvitvs, for reporting this issue. Looks like, I missed the creation of the properties, when modifying @mvieth's original PR. I'm sorry for that. I filed a PR to fix: #1593. Could you please validate it?

@mvieth: You seem to report two other/additional? issues. When do you observe these? I cannot reproduce them.

mvieth, 2021-03-04:
@rhaschke I can say that your PR fixes the parameter/property problem (at least for me).
The second issue (with `QObject::connect: No such slot`) does not appear any more, so that is fine.
The third issue (`symbol lookup error`) still appears with your PR, but after further testing apparently only with `TwistStampedDisplay` and `AccelStampedDisplay`. It appears when an incoming message is processed, e.g. when a `TwistStampedDisplay` displays messages from topic `/twist_test` and then you publish a message like so:
```
rostopic pub /twist_test geometry_msgs/TwistStamped "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: 'map'
twist:
  linear:
    x: 1.0
    y: 0.0
    z: 0.0
  angular:
    x: 1.0
    y: 0.0
    z: 0.0"
```
It seems like removing `Q_OBJECT` from `AccelStampedDisplay` and `TwistStampedDisplay` fixes this issue (`WrenchStampedDisplay` has no `Q_OBJECT`), but I don't know if that is a good fix.

rhaschke, 2021-03-04:
Thanks for the more detailed explanation. Indeed the presence of `Q_OBJECT` somehow avoid instantiation of the template instances.

Yeshasvitvs, 2021-03-04:
>  I filed a PR to fix: #1593. Could you please validate it?

@rhaschke I tested your patch, LGTM! Thanks. 

https://user-images.githubusercontent.com/6505998/109958375-7314d700-7ce6-11eb-8c53-db3047248ff1.mp4

Feel free to close the issue

rhaschke, 2021-03-04:
Closed via #1593.

PR #1593 (2021-03-04): ScrewDisplay: fix creation of properties
Fixes #1592.
closed 2021-03-04 by rhaschke