ros-visualization/rviz #1594: MarkerArray Display does not show again when toggling namespace.

Levi-Armstrong, 2021-03-10:
Currently if you disable a namespace within the marker array display it currently deletes all markers in that namespace. This prevents it from being able to toggle on again to show the namespace markers. The expected behavior was that you can toggle the namespace and it would hide and show markers within the namespace not delete them. Is the current behavior correct?

https://github.com/ros-visualization/rviz/blob/7383000e722d581dc8657f2bf6016fca88c4aee0/src/rviz/default_plugin/marker_display.cpp#L463-L466

rhaschke, 2021-03-11:
Please see [this discourse discussion](https://discourse.ros.org/t/changing-rviz-behavior-toggling-things-on-and-off/18894) on the topic. The current behavior is/was intended, but I can see value in the other behavior as well.

Levi-Armstrong, 2021-03-11:
Thank you for the information.

closed 2021-03-11 by Levi-Armstrong