ros-visualization/rviz #1595: Segfault in rviz::YamlConfigReader::readYamlNode

robin-moss, 2021-03-16:
Describe your issue here and explain how to reproduce it.

When I try to run RViz I get a segfault, running it through gdb I get:

```
(gdb) run
Starting program: /opt/ros/noetic/bin/rviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeda6f700 (LWP 26371)]
[New Thread 0x7fffe7fff700 (LWP 26372)]
[New Thread 0x7fffe77fe700 (LWP 26373)]

(rviz:26367): Gtk-WARNING **: 15:30:00.312: Theme parsing error: gtk.css:1090:0: Expected a valid selector
[New Thread 0x7fffe6f7b700 (LWP 26374)]
[1615908600.350117923]: [${node}] [ INFO] rviz version 1.14.5
[1615908600.350156173]: [${node}] [ INFO] compiled against Qt version 5.12.8
[1615908600.350165195]: [${node}] [ INFO] compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffe5f16700 (LWP 26377)]
[New Thread 0x7fffe5715700 (LWP 26378)]
[New Thread 0x7fffe4f14700 (LWP 26379)]
[New Thread 0x7fffcffff700 (LWP 26380)]
[1615908600.363080641]: [/rviz_1615908600350246943] [ INFO] Forcing OpenGl version 0.

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00007ffff7f4f5d8 in rviz::YamlConfigReader::readYamlNode(rviz::Config&, YAML::Node const&) () from /opt/ros/noetic/lib/librviz.so
```

Checking linked yaml libraries I get:

```
# ldd /opt/ros/noetic/lib/librviz.so |grep yaml
	libyaml-cpp.so.0.6 => /usr/local/lib/libyaml-cpp.so.0.6 (0x00007f753f7d4000)
```

If I attempt to build from source I get

```
[ 60%] Linking CXX executable /home/robin.moss/ros_workspaces/rviz/devel/lib/rviz/line_edit_with_button_test
/usr/bin/ld: /home/robin.moss/ros_workspaces/rviz/devel/lib/librviz.so: undefined reference to `YAML::detail::memory::create_node()'
/usr/bin/ld: /home/robin.moss/ros_workspaces/rviz/devel/lib/librviz.so: undefined reference to `YAML::Emitter::Write(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
...
```
and so on

checking `ldconfig` gets me

```
$ ldconfig -p | grep yaml
	libyaml-0.so.2 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libyaml-0.so.2
	libyaml-cpp.so.0.6 (libc6,x86-64) => /usr/local/lib/libyaml-cpp.so.0.6
	libyaml-cpp.so.0.6 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.6
	libyaml-cpp.so (libc6,x86-64) => /usr/local/lib/libyaml-cpp.so
	libyaml-cpp.so (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libyaml-cpp.so
```

I've run out of ideas :( any help would be appreciated 

### Your environment
* OS Version: e.g. Ubuntu 20.04
* ROS Distro: [Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
    ```
rviz version 1.14.5
compiled against Qt version 5.12.8
compiled against OGRE version 1.9.0 (Ghadamon)
    ```
* If source build, which git commit? Noetic-devel

rhaschke, 2021-03-16:
The correct libyaml-cpp version is the one in /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.6. The one in `/usr/local` probably conflicts with it. See #1591 as well, which had exactly the same issue.

closed 2021-03-16 by rhaschke

robin-moss, 2021-03-17:
Thanks for the fast reply, I had looked at the other ticket but everything looked fine, obviously I missed the `/usr/local` part :facepalm: 

Removing that has fixed my issue thanks again fo a fast reply