ros-visualization/rviz #1599: EditableEnumProperty hide bottom part of character in edit status

mviz-mdc, 2021-03-25:
There is a UI issue for EditableEnumProperty.  When it enter edit status, It will hide the bottom part of 'g', 'q', 'y' etc. It make a little confusing.
[image]

If modify code here, it could be solve.
[image]

Effect after modification:
[image]

If there is any better way to fix it, maybe we could have a discuss here.

### Environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1616568586.970028200]: rviz version 1.13.15
[ INFO] [1616568586.970076100]: compiled against Qt version 5.9.5
[ INFO] [1616568586.970083600]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1616568586.973661300]: Forcing OpenGl version 0.
[ INFO] [1616568587.067990800]: Stereo is NOT SUPPORTED
[ INFO] [1616568587.068078500]: OpenGl version: 3.1 (GLSL 1.4).
```

rhaschke, 2021-03-25:
Thanks for this issue report and the proposed solution. Do you mind filing an appropriate pull request? Thanks.

rhaschke, 2021-03-25:
Fix applied in b3ce39751df614999039359bf6cb646577cbbae3. Thanks again.

closed 2021-03-25 by rhaschke