ros-visualization/rviz #1605: Add new imagepanel, but it doesn't display correctly.

jeffryHo, 2021-04-01:
When I add imagepanel, there is always a display error. I tried many ways, but I couldn't fix the problem. Can you fix it?

[image]

jeffryHo, 2021-04-01:
redo

rhaschke, 2021-04-01:
This works without problems on my side. Did you configure an image topic to read from?
[image]

This question should be better asked at https://answers.ros.org. Please, post real issues and bug reports here.

jeffryHo, 2021-04-02:
It's a problem that's probabilistically recurring. You can try to add imagepanel few more times, and the problem will recur, and you don't need to select any topics.

jeffryHo, 2021-04-02:
I can probably locate the problem now, and I'll test it for a long time to make sure it's resolved.

rhaschke, 2021-04-02:
Obviously, I missed the actual problem. Could you please more explicitly explain your problem?

jeffryHo, 2021-04-06:
Environment:
* OS Version: Ubuntu 18.04
* ROS Distro: [Melodic]
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
[ INFO] [1617672134.333878300]: rviz version 1.13.16
[ INFO] [1617672134.333933900]: compiled against Qt version 5.9.5
[ INFO] [1617672134.333944200]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617672134.339914900]: Forcing OpenGl version 0.
[ INFO] [1617672134.545871700]: Stereo is NOT SUPPORTED
[ INFO] [1617672134.546045100]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
[ INFO] [1617672134.546104200]: OpenGl version: 3.1 (GLSL 1.4).

Operation procedure:
1.open rviz 
2.add imagepanel from displays panel 
3. add imagepanel multiple times 

Expect result:
imagepanel show defalut texture image correctly

Actual results:
some imagepanel is no show it correctly

Location Information
setContentWidget and addDocWidget will reset objects parent, and that cause resize event unnecessary sended.
render widget responded resize events twices, one of which was wrong.

[image]

fix way：
```C++
 PanelDockWidget* dock;
  
  dock = new PanelDockWidget(name);

  dock->setFloating(floating);
  dock->setObjectName(name); // QMainWindow::saveState() needs objectName to be set.
  addDockWidget(area, dock);

  dock->setContentWidget(panel);
```

By the way, what dose dock->setFloating(floating) use for?  it run befor addDockWiget doesn't seem to do much here.

rhaschke, 2021-04-06:
If I understand correctly, you suggest to change the order of calls: First call addDockWidget() and then setContentWidget(), which will save a call to `resize()`. Could you please file a PR for this change?
The question remains, why the code cannot handle two resize events in a row. What do you mean by "one of those resize events was _wrong_"? Just superfluous or badly parameterized?

jeffryHo, 2021-04-07:
yes , your understanding is right. 

I add a imagepanel from displaypanel.
if the imagepanel display correctly, the image is like

[image]

 I add log print in QtOgreRenderWindow ::resizeEvent and it print log is

[ INFO] [1617766276.440939400]: rviz version 1.13.16
[ INFO] [1617766276.440979000]: compiled against Qt version 5.9.5
[ INFO] [1617766276.440985800]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617766276.445404100]: Forcing OpenGl version 0.
[ INFO] [1617766276.575364100]: Stereo is NOT SUPPORTED
[ INFO] [1617766276.575495500]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
[ INFO] [1617766276.575552200]: OpenGl version: 3.1 (GLSL 1.4).

-------------------------------------------------

OgreRenderWindow resize:543;688
OgreRenderWindow resize time stamp:11:31:16 685685

DisplaysPanel addNew end:11:31:21 055055

-------------------------------------------------

OgreRenderWindow resize:342;166
OgreRenderWindow resize time stamp:11:31:21 057057

but if imagepanel display image is not in the right way
it display image like 

[image]

and log print is 

[ INFO] [1617766201.796327700]: rviz version 1.13.16
[ INFO] [1617766201.796388800]: compiled against Qt version 5.9.5
[ INFO] [1617766201.796400500]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617766201.802863000]: Forcing OpenGl version 0.
[ INFO] [1617766202.045629400]: Stereo is NOT SUPPORTED
[ INFO] [1617766202.045809000]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
[ INFO] [1617766202.045882100]: OpenGl version: 3.1 (GLSL 1.4).

-------------------------------------------------

OgreRenderWindow resize:543;688
OgreRenderWindow resize time stamp:11:30:02 3131

DisplaysPanel addNew end:11:30:07 507507

-------------------------------------------------

OgreRenderWindow resize:342;166
OgreRenderWindow resize time stamp:11:30:07 508508

-------------------------------------------------

**OgreRenderWindow resize:640;480
OgreRenderWindow resize time stamp:11:30:07 528528**

There are one more resize event had happend than the normal one.
I thought that maybe the Qt event loop is no correctly, The sequence of adding widgets may be wrong.

so that I adjusted the order of widgets added. addDockWidget first , and then setContentWidget. Magically , the problem
is fixed. And dock->setFloating(floating) is working too.  the imagepanel floating above the mainwindow

[image]

well , In order to be able to keep the same behavior pattern, dock->setFloating is parameter must be false， the dockwidget will docked in the leftarea.

rhaschke, 2021-04-07:
Thanks for this clarification. Could you please file a PR with your solution? Thanks.

rhaschke, 2021-04-08:
Fixed via #1611.

closed 2021-04-08 by rhaschke

jeffryHo, 2021-04-09:
Thanks for merged my PR, I found that you changed the setting of floating, but when floating is set to true, the panel cannot be displayed because you did not add the panel to the main window. Have you ever tested it?

if floating is true then rviz show like this
[image]

When the PanelDockwidget is created, its parent class is not specified.

I don't understand what you mean "floating" mutually excludes "docked"? After addDockWidget you can also make the widow floating.