ros-visualization/rviz #1606: Whenever open a dialog, the data stop update in 3D area and panels

tsukione, 2021-04-04:
Maybe this is not an issue, just a qustion.
I found every time a open a Dialog, such as NewDisplay, NewPanels, the pointcloud and image stop updating.

I check the code, found every time before open a dialog, will call 
https://github.com/ros-visualization/rviz/blob/7383000e722d581dc8657f2bf6016fca88c4aee0/src/rviz/visualization_frame.cpp#L642
When the dialog is closed, a startUpdate() wil called.

So, I just wondering why the stopUpdate() should be called. I comment the  manager_->stopUpdate(); line, It still working perfectly, and the pointcloud and image is continuous changing, I'm happy with that :)

So, could you help me to get out from this question? thanks.

### Environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1617528442.570476806]: rviz version 1.13.16
[ INFO] [1617528442.570531904]: compiled against Qt version 5.9.5
[ INFO] [1617528442.570563003]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617528442.574762879]: Forcing OpenGl version 0.
[ INFO] [1617528442.680312957]: Stereo is NOT SUPPORTED
[ INFO] [1617528442.680650647]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
[ INFO] [1617528442.680780743]: OpenGl version: 3.1 (GLSL 1.4).
 ```

rhaschke, 2021-04-05:
This protection was introduced in #631 on purpose. Did you try what happens if you overload the main rendering loop?

tsukione, 2021-04-06:
> This protection was introduced in #631 on purpose. Did you try what happens if you overload the main rendering loop?

Got it, I will test that later

rhaschke, 2021-06-18:
Reverting #631 doesn't prevent message processing and display updating from pausing. This is due to the fact that all dialogs are _modal_, i.e. prevent Qt's event processing for other windows.

tsukione, 2021-06-18:
I just tested it and commented out stopUpdate, and the data was still refreshing when the dialog was opened. So it has nothing to do with the modal window.

rhaschke, 2021-06-18:
I test it as well (see [this branch](https://github.com/rhaschke/rviz/tree/cont-message-processing)) and the display wasn't refreshing for me!? Could you provide your branch as well for testing?

tsukione, 2021-06-18:
I pulled up the latest code(melodi-devel), and only commet this line:
https://github.com/ros-visualization/rviz/blob/3fa78da828e3740c0e92f3e8f7dbc49d283422d8/src/rviz/visualization_manager.cpp#L306

tsukione, 2021-06-18:
I have download your branch and tested, the point cloud data is still updating in the 3D area.

rhaschke, 2021-06-18:
Hm. Don't know what happened before. But it works now for me as well. In principle, I'm fine to accept this behaviour for Noetic.

closed 2021-06-19 by rhaschke

rhaschke, 2021-08-22:
@tsukione, unfortunately, your feature request caused a regression: https://github.com/ros-visualization/rviz/issues/1641.
I'm tempted to revert to the previous state.

PR #1639 (2021-06-19): Continue processing messages while displaying dialogues
Partially reverts #631. Fixes #1606.
commit 227678f0 (2021-06-19): Continue processing messages while displaying dialogues (#1639)

Partially reverts #631. Fixes #1606.