ros-visualization/rviz #1610: About the resize of the image window

HankWong-guangdong, 2021-04-08:
https://github.com/ros-visualization/rviz/blob/ba536b2d5baae4be167f9741a3f5bcb05d129af6/src/rviz/default_plugin/image_display.h#L89
[image]

In the image window, the drag on the lower left corner and the lower right corner to resize the window is more difficult  than the drag on the upper left corner  and  upper right corner。I find that the class RenderPanel override the mouseMoveEvent 、mousePressEvent and mouseReleaseEvent。But only the render panel in the central of the main window responds to mouseEvent, the render area of the Image window  dose not respond to mouseEvent. For the image window, why not write a new class to inherit the class RenderPanel and override the  mouseMoveEvent 、mousePressEvent  and mouseReleaseEvent. In all the three functions, use the corresponding functions QWidget::mouseMoveEvent 、QWidget::mousePressEvent 、and QWidget::mouseReleaseEvent  in QWidget to process the mouseEvent.  That will resolve the dragging issues.

rhaschke, 2021-04-08:
I agree that the RenderPanel used for the `ImageDisplay` doesn't need its specific mouse handling in `onRenderWindowMouseEvents()`. However, I'm not convinced yet that resorting to the default QWidget behavior will resolve the described issue. I think, the root cause are very thin window borders.
Nevertheless, you are very welcome to file a PR implementing a satisfactory solution.

HankWong-guangdong, 2021-04-08:
> I agree that the RenderPanel used for the `ImageDisplay` doesn't need its specific mouse handling in `onRenderWindowMouseEvents()`. However, I'm not convinced yet that resorting to the default [...]

I have tried to create a new class inheriting  the class RenderPanel  for the image window. In the new class, I override the mouseMoveEvent 、mousePressEvent and mouseReleaseEvent. In  all the three functions, I use the corresponding functions QWidget::mouseMoveEvent 、QWidget::mousePressEvent 、and QWidget::mouseReleaseEvent in QWidget to process the mouseEvent. Then, the drag on the lower left corner and the lower right corner to resize the window becomes as convenient as the drag on the upper left corner and upper right corner。You can have a try。

rhaschke, 2021-04-08:
Good. If you already implemented this, please file a corresponding PR.