ros-visualization/rviz #1612: RViz may crash when duplicating the Image Display

sunzbllbz, 2021-04-10:
When I try to duplicate several Image Displays in the Displays Panel, I suddenly press the "Remove" button in the Displays Panel for some reasons. Then I found that RViz crashed after my operations. 
I try to find the reason with GDB and finally I got the call stack of RViz. The call stack is shown as below.
```
 Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x0000000800000008 in ?? ()
(gdb) bt
#0  0x0000000800000008 in  ()
#1  0x00007f14a197e142 in rviz::DisplaysPanel::onDuplicateDisplay() ()
    at /opt/ros/melodic/lib/librviz.so
#2  0x00007f14a0bb4555 in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007f14a128bba2 in QAbstractButton::clicked(bool) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007f14a128bdba in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007f14a128d19a in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007f14a128d38d in QAbstractButton::mouseReleaseEvent(QMouseEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007f14a11d9048 in QWidget::event(QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007f14a119a83c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007f14a11a265f in QApplication::notify(QObject*, QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007f14a0b858d8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007f14a11a1632 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007f14a11f416b in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
---Type <return> to continue, or q <return> to quit---
#13 0x00007f14a11f67da in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007f14a119a83c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007f14a11a2104 in QApplication::notify(QObject*, QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007f14a0b858d8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007f149b516583 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#18 0x00007f149b518055 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#19 0x00007f149b4ef2eb in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#20 0x00007f1489f51260 in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#21 0x00007f1499f21537 in g_main_context_dispatch ()
    at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007f1499f21770 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007f1499f217fc in g_main_context_iteration ()
    at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007f14a0bde88f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#25 0x00007f14a0b8390a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>---Type <return> to continue, or q <return> to quit---
) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007f14a0b8c9b4 in QCoreApplication::exec() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00005628fbd90dcf in main ()

```
I was confused about this because I press "Duplicate" button firstly, but it seemed that RViz removed the Image Displays which was being copied and then continued to duplicate the Image Displays which had been deleted. Then I do another experiment and got the same result that RViz crashed. In this experiment, I selected more Image Displays to duplicate (more than 50). I pressed the "Duplicate" button on the Displays Panel twice quickly and then I pressed the "Remove" button. RViz crashed and I got the call stack as below.
```
Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x000000000000001e in ?? ()
(gdb) bt
#0  0x000000000000001e in  ()
#1  0x00007f1305b2c142 in rviz::DisplaysPanel::onDuplicateDisplay() ()
    at /opt/ros/melodic/lib/librviz.so
#2  0x00007f1304d62555 in QMetaObject::activate(QObject*, int, int, void**) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007f1305439ba2 in QAbstractButton::clicked(bool) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007f1305439dba in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007f130543b19a in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007f130543b38d in QAbstractButton::mouseReleaseEvent(QMouseEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007f1305387048 in QWidget::event(QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007f130534883c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007f130535065f in QApplication::notify(QObject*, QEvent*) ()
[... 73 lines cut ...]
#58 0x00007f1304d3a9b4 in QCoreApplication::exec() ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#59 0x000056263ccfbdcf in main ()

```
According to the call stack, I found that when RViz created an Image Display, it would construct a RenderWidget, which would call QApplication::sync() (the 25th line in the call stack). After QApplication::sync() was called, RViz would process the Qt events, as well as removing the display plugins. After RViz finished to remove the Image Displays, it would go back to rviz::DisplaysPanel::onDuplicateDisplay() and continued to duplicate the Image Displays. However, the Image Displays had beed removed and RViz would visited the null pointer, which would result in crash.
I confuse about why QApplication::sync() is called in the constructor of rviz::RenderWidget. Can QApplication::sync() be removed from the constructor?
If the QApplication::sync() is not able to be removed, I think we should take some other measures to solve this problem. To avoid this problem, one of the solutions is adding a QProcessDiag to the rviz::DisplaysPanel::onDuplicateDisplay() as below. While duplicating the displays, a QProcessDiag will be shown on the RViz, which is modal. Therefore, user will be not able to press the "Remove" button and the crash will be avoided. The codes about the diag are as below.
[image]
And the effect picture is also as below.
[image]
This is just a suggestion. If there are more reasonable solution, we can make a further discussion here.

### Your environment
* OS Version: Ubuntu 18.04
* ROS Distro: Melodic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
[ INFO] [1617852872.803608900]: rviz version 1.13.15
[ INFO] [1617852872.803677500]: compiled against Qt version 5.9.5
[ INFO] [1617852872.803696100]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1617852872.808645600]: Forcing OpenGl version 0.
[ INFO] [1617852873.113495600]: Stereo is NOT SUPPORTED
[ INFO] [1617852873.113581400]: OpenGl version: 3.1 (GLSL 1.4).
```

rhaschke, 2021-04-10:
Thanks for this detailed analysis. `QApplication::sync()` was introduced by @rlober in #1263. I'm not sure why it was needed. @rlober, do you remember?
https://github.com/ros-visualization/rviz/blob/3cdc03a5186a53f644c5abd55b9971457bdffa1c/src/rviz/ogre_helpers/render_widget.cpp#L78-L82

In any case, I like your approach to display a progress dialog. Do you mind filing a corresponding PR?

tsukione, 2021-04-11:
What a coincidence, I encounter an issue caused by the QApplication::sync() too, the sync here give a huge hidden danger in initialize. sync() force Qt process events, and that will makes qt to call some method, such as getViewData(), even the onIntialize() is not finished. If I access some member initialize after render_panel_ in getViewData(), the program is core dumped. Maybe call sync() in constructor is not a wise approach.
[image]

console output:
```
[ INFO] [1618104707.573993981]: rviz version 1.13.17
[ INFO] [1618104707.574048878]: compiled against Qt version 5.9.5
[ INFO] [1618104707.574082675]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1618104707.577835020]: Forcing OpenGl version 0.
[ INFO] [1618104707.674521931]: Stereo is NOT SUPPORTED
[ INFO] [1618104707.674754515]: OpenGL device: llvmpipe (LLVM 10.0.0, 256 bits)
[ INFO] [1618104707.674825411]: OpenGl version: 3.1 (GLSL 1.4).
ImageDisplay::getViewData is accessed when ImageDisplay is not initialized.

```

tsukione, 2021-04-15:
hi @rhaschke, It seems @rlober haven't been active for a long time. 
The official documents of Qt discourage use QApplication::sync(), and all of us can't see what kind of problem could the sync() solve. Only those problem it cause could be seen. So I propose to delete they. 
https://github.com/ros-visualization/rviz/blob/3917c01555058c6c117bd32213fb46919a9d1bb6/src/rviz/ogre_helpers/render_widget.cpp#L77-L82

rhaschke, 2021-04-25:
@tsukione, please test and review #1624.

tsukione, 2021-04-26:
Thank you accept my proposal. I have test the PR in my local environment, It works good.

rhaschke, 2021-05-14:
Fixed via #1624.

closed 2021-05-14 by rhaschke