ros-visualization/rviz #1613: RVIZ may crash when I scroll mouse in 3D view area

solookin, 2021-04-10:
Hello, I meet a rviz crash.

when I scroll mouse in 3D view area and let the view scale very very small or very very big , the rivz will crash.

step 1: place a Plant Flat in 3D view area and set view type "TopDownOrtho".

[image]

step 2: scroll my mouse in 3D view area, when the view scale is very very  big or very very small, rviz will crash.
[image]

when crash, i can see crash log in terminal.

[image]

Your environment
  OS Version: Ubuntu 18.04
  ROS Distro: Melodic
  RViz, Qt, OGRE, OpenGl version as printed by rviz:

rhaschke, 2021-04-10:
I'am afraid there is no generic min/max value for the scale. People might use very different scene scales as well...

solookin, 2021-04-12:
> I'am afraid there is no generic min/max value for the scale. People might use very different scene scales as well...

OK. thanks.

closed 2021-04-12 by rhaschke

solookin, 2021-04-13:
Although people might use very different scene scales, but we can do something to avoid Rviz crashes ?

I edit the code in src/rviz/default_plugin/view_controllers/fixed_orientation_ortho_view_controller.cpp, it may work.

```
void FixedOrientationOrthoViewController::updateCamera()
{
  orientCamera();

  float width = camera_->getViewport()->getActualWidth();
  float height = camera_->getViewport()->getActualHeight();

  float scale = scale_property_->getFloat();
  Ogre::Matrix4 proj;
  
  // start
  // keep scale always valid to avoid rviz crashes #1613
  if (std::abs(width / scale / 2) == INFINITY || std::abs(height / scale / 2) == INFINITY) {
    scale_property_->setValue(last_valid_scale_);
    return;
  }
  if (std::abs(width / scale / 2) < 0.000001 || std::abs(height / scale / 2) < 0.000001) {
    scale_property_->setValue(last_valid_scale_);
    return;
  }
  last_valid_scale_ = scale;
  
  // end
  
  
  buildScaledOrthoMatrix( proj, -width / scale / 2, width / scale / 2, -height / scale / 2, height / scale / 2,
                          camera_->getNearClipDistance(), camera_->getFarClipDistance() );
  camera_->setCustomProjectionMatrix(true, proj);

  // For Z, we use half of the far-clip distance set in
  // selection_context.cpp, so that the shader program which computes
  // depth can see equal distances above and below the Z=0 plane.
  camera_->setPosition( x_property_->getFloat(), y_property_->getFloat(), 500 );
}
```

rhaschke, 2021-04-13:
Please file a PR with your proposed changes. Here, I cannot immediately see what you actually propose to change.