ros-visualization/rviz #1621: DepthCloud memory leak

christian-rauch, 2021-04-15:
I am using the `DepthCloud` display type to back-project colour and depth to a 3D point cloud in RViz. When I click the `Add` button I can add a new display type. While this new pop-up window is shown, the point cloud in the 3D view is not updated and the memory consumption by RViz rises until its process gets killed by `oom-kill`.

Reproduce:
1. play a bag file with depth and colour images
2. open RViz
3. add and configure the `DepthCloud` display type
4. try to add a new display type by clicking on the `Add` button

My configuration:
[image]

Memeory consumption of the process:
[image]

### Your environment
* OS Version: Ubuntu 20.04
* ROS Distro: Noetic
* RViz, Qt, OGRE, OpenGl version as printed by rviz:
```
rviz version 1.14.5
compiled against Qt version 5.12.8
compiled against OGRE version 1.9.0 (Ghadamon)
Forcing OpenGl version 0.
Stereo is NOT SUPPORTED
OpenGL device: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
OpenGl version: 4.6 (GLSL 4.6) limited to GLSL 1.4 on Mesa system.
```

rhaschke, 2021-04-15:
This is related to #1606. Stopping data processing as introduced in #631 will result in the accumulation of ROS messages in the callback queue. However, this should only happen for large queue sizes (and yours is 1)!?
https://github.com/ros-visualization/rviz/issues/868#issuecomment-120614852 suggests that there is another cause for that issue.

rhaschke, 2021-06-24:
Fixed via #1635

PR #1635 (2021-06-19): Pause asynchronous ROS updates with synchronous ones
VisualizationManager uses `[start|stop]Update()` to pause ROS updates during dialog box interaction (#631).
However, asynchronous updates via `threaded_queue_` were still performed, leading to a memory leak (#1621),
because messages were partially processed by the thread (i.e. kept in memory), but not finally processed
(and cleaned up) by the synchronous `Display::update()` call.
Pausing threaded processing too resolves that issue.

Fixes #1621. Fixes #868.
@simonschmeisser: Do you see any issue pausing asynchronous message processing too wh [... cut]
closed 2021-06-24 by rhaschke